sidoh/esp8266_milight_hub #203: Alexa -> SmartThings -> ESPMH setting color bulbs to white

MisterC925, 2018-01-12:
When controlled from smartthings (via fireboy1919's integration) I can set a bulb to a color and then back to white, however, when I add alexa into the mix, instead of sending set_white like when I press the white button, it sends set color 0 (or set hue 0) which does not set the bulbs to white.

I'm not sure where in the communications chain this is breaking, but I was curious if this is a case that can be handed in a firmware update to the ESPMH or should I look more closely at the smartthings integration?

Thanks!

MisterC925, 2018-01-12:
2018-01-11 11:55:04.057 PM PST 					13 hours ago | DEVICE |   | color | 0.0 |   | Bedside White Light color is 0.0
-- | -- | -- | -- | -- | -- | --

sidoh, 2018-01-12:
What button is causing the color to get set to 0?  This does not sound like a problem with ESPMH.  Sounds like something upstream is misbehaving.

MisterC925, 2018-01-13:
I think that's what I'm trying to figure out, I'll do some more debugging tonight.

It's either Alexa telling smartthings to set color to 0 or somehow the smartthings's app is sending color 0 instead of set_white.   I wasn't sure if setting color to 0 for white was expected behavior or not.

Thanks!

MisterC925, 2018-01-15:
Okay so I've dug further and the problem is definately between Alexa and Smartthings.

I'm going to try adding an exception to the smartthings app code to send set_white when color is 0.

closed 2018-01-15 by MisterC925