sidoh/esp8266_milight_hub #204: Support for status LED?

marrold, 2018-01-16:
Hi, is there any support for a status LED? (Power, connected to Wifi, sending command via RF etc? 

I had a look at the code and nothing jumped out at me.

If there's no code for a status LED yet I will try to add it.

Thanks

sidoh, 2018-01-16:
Nope, but sounds like a good idea.

cmidgley, 2018-02-16:
I've implemented a status LED feature - see this [Pull Request](https://github.com/sidoh/esp8266_milight_hub/pull/229) for the changes.  It adds support for both the build-in LED (pin 2) and using any other pin (set pin in the UI), and flash codes for wifi setup, wifi working, and packet transmission.  Pretty easy to add others, but too many unique flash codes gets confusing!  Check out the readme for details.

closed 2018-05-05 by sidoh