sidoh/esp8266_milight_hub #206: Please add a sponsor link :)

fliphess, 2018-01-18:
Hey @sidoh 

Having so much joy of this project almost every day of the week (You somehow made it fun to turn the lights on and off and on and off... :)), I would love to send some beers or make a small donation to show some gratitude and support this project and I think there are plenty other users feeling the same way. 

I can imagine that you and some frequent committers from time to time need some equipment (IE a newer model milight bulb for testing, a new esp8266 or some other compatible pcb or a nice Belgian beer on a coding night :))

As an extremely happy and grateful user I would love to support the project and the people working on it.... 

So..... How about adding a flattr/pledgie/bitcoin address/some other open source funding platform? :)

sidoh, 2018-01-19:
Hi @fliphess,

It makes me really happy to hear the project has added value for you.  Thank you for the kind words and your generosity. 

If generous folks like yourself are open to a charitable donation, that'd certainly make me very happy.  Here are some whose mission I support, but obviously any organization that you feel is doing good in the world is great too.

* [Water.org](https://www.water.org)
* [Brain & Behavior Research Foundation](https://www.bbrfoundation.org/)
* [Electronic Frontier Foundation](https://www.eff.org/)

I'll add some links and a blurb to the README sometime soon.  Thank you for the suggestion.

sidoh, 2018-05-31:
Sorry this took me so long to do: https://github.com/sidoh/esp8266_milight_hub/commit/758e1d05422a35ee8d1dd528e7fda4fca65da8c9

Thanks again.  Means a lot to hear that it's brought you some happiness :)

closed 2018-05-31 by sidoh