sidoh/esp8266_milight_hub #207: Black bulb or color picker in Hass

MimbaMonkeyHouse, 2018-01-19:
Hi Chris,

Thanks for the work you did on this.

One issue I have is that I am currently unable to have both the color picker and the light icons reflecting the actual color of the bulb in Home Assistant (v60.1). 
Right now, I can get the icons to change color by using the computed_color: true command or the color picker by using the color: true or rgb:true. 
If using both computed_color: true and one of the other 2 options at the same time, the color picker is on, but the icon is black (R:0, G:0, B:0)...

sidoh, 2018-01-19:
It sounds like you're changing your HASS config, which shouldn't be necessary.  Select the "computed_color" field in the ESPMH config.

MimbaMonkeyHouse, 2018-01-19:
Thanks for getting back.
Yes, I did. With the rgb: true setting in Home Assistant, there is no change in bulb color in HA .60.1! What setting should be used in HA?
The ESPMH config is set up with just computed_color...

sidoh, 2018-01-21:
It was working for me with HASS 0.59.  I just upgraded to 0.61.1, and it's still working.

This is my ESPMH state field config:

[image]

The relevant section of my HASS config is in the wiki for this project:

https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant

MimbaMonkeyHouse, 2018-01-24:
Hi Chris,

Thanks for getting back. I proof checked all settings in both ESPMH (& updated to 1.6.5 nomcu v2) and HASS (v60.1). All is the same as your and here are relevant parts of my lights.yaml for my RGBW, RGB_CCT and CCT respectively:
  state_topic: milight/updates/0x22CB/rgbw/1
  <<: &MILIGHTRGBW_PARAMS
      platform: mqtt_json
      color_temp: false
      rgb: true
      brightness: true
  state_topic: milight/updates/0x224C/rgb_cct/1
  <<: &MILIGHT_PARAMS
      platform: mqtt_json
      rgb: true
      color_temp: true
      brightness: true
  state_topic: milight/updates/0x314/cct/1
  <<: &MILIGHTCCT_PARAMS
      platform: mqtt_json
      color_temp: true
      rgb: false
      brightness: true
      qos: 1
      retain: true
The icon color in HA updates for both RGBW and CCT, but not for RGB-CCT. As explained earlier, if rgb: false and computed_color: true are applied, then the icon color updates, but no color picked is available.

Do you have a solution for RGB-CCT? Should I update GitHub with this so others can ring in?

Cheers

sidoh, 2018-01-29:
The `rgb` key in the HASS config should not be changed.  It should be set to true.  `computed_color` is an ESPMH config, not HASS.  My HASS section for RGB+CCT is:

```yaml
    state_topic: milight/updates/0x1/rgb_cct/1

    # Use a YAML anchor for common settings so we can just reference
    # them for other lights.
    <<: &MILIGHT_PARAMS
      platform: mqtt_json
      color_temp: true
      rgb: true
      brightness: true
```

JSmid6, 2018-01-29:
I'm having the same issue with my rgbw bulbs, the cct bulbs work fine.

sidoh, 2018-01-30:
Both rgbw and rgb+cct work for me. 

[image]

[image]

Please copy/paste your HASS configs and a settings dump from ESPMH (just navigate to http://milight-hub.local/settings), making sure to censor any passwords.

garmck, 2018-01-30:
Just confirming I **do not** have this issue either - running 0.61.1

MimbaMonkeyHouse, 2018-01-30:
Thanks for taking the time to reply. 

Mmh, then something is going strangely with my setup. I’ve been getting this with both v60.1, v61.1 and now v62.0… I’m running HA on a Synology docker. I am currently using some of the bulbs with the Milight integration as my RF gateways do not reach far enough to get all lights covered. I use 2 RF gateways. Settings dumps are attached along with my lights.yaml configuration. 

This is leaving me perplexed and I may be missing something obvious… so thanks again!

Cedric

MimbaMonkeyHouse, 2018-01-30:
I thought you may want to see a screenshot of what I get in HA so here they are. You’ll see the RGB icons are also blacked out when on…

JSmid6, 2018-01-30:
I changed the settings around and I no longer have this issue. I used the setting as chris mentioned above.

sidoh, 2018-01-30:
@MimbaMonkeyHouse, if you're attaching things to your messages (via email, it looks like?), they're not coming through.

MimbaMonkeyHouse, 2018-01-30:
Ok, sorry about that. Guess the monkey part of name stands true...

Here are the codes. ESPMH Gateway 1:
`{"admin_username":"EDIT","admin_password":"EDIT","ce_pin":16,"csn_pin":15,"reset_pin":0,"radio_interface_type":"nRF24","packet_repeats":100,"http_repeat_factor":1,"auto_restart_period":1200,"mqtt_server":"EDIT","mqtt_username":"EDIT","mqtt_password":"EDIT","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/state/:device_id/:device_type/:group_id","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":2,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"device_ids":[4369,796],"gateway_configs":[[8780,8899,6],[8907,8899,6],[925,8899,6],[796,8899,6],[788,8899,6],[8541,8899,6],[8668,8899,6],[8660,8899,6],[2143,8899,6],[2270,8899,6],[2262,8899,6],[1617,8899,5],[1744,8899,6],[1736,8899,5],[5121,8899,6],[5248,8899,6],[327,8899,6]],"group_state_fields":["state","brightness","color_temp","computed_color"]}`

ESPMH Gateway 2:
`{"admin_username":"EDIT","admin_password":"EDIT","ce_pin":16,"csn_pin":15,"reset_pin":0,"radio_interface_type":"nRF24","packet_repeats":100,"http_repeat_factor":1,"auto_restart_period":1200,"mqtt_server":"EDIT","mqtt_username":"EDIT","mqtt_password":"EDIT","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":2,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":8,"device_ids":[8541,8780,796,925],"gateway_configs":[[8780,8899,6],[8907,8899,6],[925,8899,6],[796,8899,6],[788,8899,6],[8541,8899,6],[8668,8899,6],[8660,8899,6],[2143,8899,6],[2270,8899,6],[2262,8899,6],[1617,8899,6],[1744,8899,6],[1736,8899,6],[5121,8899,6],[5248,8899,6],[327,8899,6],[4369,8899,5],[0,0,5]],"group_state_fields":["state","brightness","color_temp","computed_color"]}`

Lights.yaml:
`# / Upstairs
- name: "Stairs"
  command_topic: milight/0x224C/rgb_cct/1
  state_topic: milight/updates/0x224C/rgb_cct/1
  <<: &MILIGHT_PARAMS
      platform: mqtt_json
      rgb: true
      color_temp: true
      brightness: true
- name: "Upstairs Stairs"
  command_topic: milight/0x224C/rgb_cct/1
  state_topic: milight/updates/0x224C/rgb_cct/1
  <<: *MILIGHT_PARAMS #copy of light.stairs created for floorplan multilayers
- name: "Upstairs Sofa"
  command_topic: milight/0x224C/rgb_cct/2
  state_topic: milight/updates/0x224C/rgb_cct/2
  <<: *MILIGHT_PARAMS
- name: "Upstairs Bed"
  command_topic: milight/0x224C/rgb_cct/3
  state_topic: milight/updates/0x224C/rgb_cct/3
  <<: *MILIGHT_PARAMS
- name: "Upstairs Toilet"
  command_topic: milight/0x55'55/cct/1
  state_topic: milight/updates/0x5555/cct/1
  <<: &MILIGHTCCT_PARAMS
      platform: mqtt_json
      color_temp: true
      rgb: false
      brightness: true
# Upstairs /

# Electric Room
- name: "Electric Room RGB"
  command_topic: milight/0x22CB/rgbw/1
  state_topic: milight/updates/0x22CB/rgbw/1
  <<: &MILIGHTRGBW_PARAMS
      platform: mqtt_json
      color_temp: false
      rgb: true
      brightness: true
# Electric Room /

# Garden
- platform: mqtt
  name: "Pool Light"
  state_topic: "stat/th16_/POWER"
  command_topic: "cmnd/th16_2/POWER"
  payload_on: "ON"
  payload_off: "OFF"
- name: "Spots"
  command_topic: milight/0x39D/rgb_cct/1
  state_topic: milight/updates/0x39D/rgb_cct/1
  <<: *MILIGHT_PARAMS
- name: "Patio Strip"
  command_topic: milight/0x39D/rgb_cct/2
  state_topic: milight/updates/0x39D/rgb_cct/2
  <<: *MILIGHT_PARAMS
- name: "Bale Top"
  command_topic: milight/0x39D/rgb_cct/3
  state_topic: milight/updates/0x39D/rgb_cct/3
  <<: *MILIGHT_PARAMS
[... 118 lines cut ...]
# Kids /

# / Street
- name: "Street 1"
  command_topic: milight/0x314/cct/1
  state_topic: milight/updates/0x314/cct/1
  <<: *MILIGHTCCT_PARAMS
- name: "Street 2"
  command_topic: milight/0x314/cct/2
  state_topic: milight/updates/0x314/cct/2
  <<: *MILIGHTCCT_PARAMS
- name: "Street RGB"
  command_topic: milight/0x31C/rgbw/1
  state_topic: milight/updates/0x31C/rgbw/1
  <<: *MILIGHTRGBW_PARAMS
# Street /
`

I've rid the codes of optional entries like Effects, QOS or Retain in an effort to troubleshoot.

And the actual HA interface rendering:
[image]
[image]

Thanks to all the participants in this thread, I hope we find my mistake.

MimbaMonkeyHouse, 2018-01-30:
Well, that was easy! This solved my issue. I was sure I read somewhere that updates should now be used as a topic instead of state.
Thanks to @gramck for enacting a quick solution and to @sidoh for a great integration... Maybe updating the docs could help. Let me know if help is needed.
[image]

sidoh, 2018-01-31:
Thanks, @garmck!  Good catch.

@MimbaMonkeyHouse, you're right, the wiki page was outdated.  I updated it.

There's more info on the difference between the update and state topics here:

https://github.com/sidoh/esp8266_milight_hub#updates

sidoh, 2018-01-31:
Sounds like this is resolved, but please feel free to reopen if not.

closed 2018-01-31 by sidoh