sidoh/esp8266_milight_hub #208: NRF24L01 Power settings

Markus1970, 2018-01-19:
Hi,
first of all many many thanks for this great work.
Would it be possible to add on the WebIf any possibilty to set the power Settings when NRF24L01 amplified modules are used? For example I use a NRF24L01 +lna +pa attached to a Wemos D1 mini. Sniffing is working fine but when send a command the power goes down completly. A Setting were the power settings for the Radio can be defined were great !!

Many thanks

Markus

sidoh, 2018-01-19:
Think a setting makes sense.  It'll likely be a while before I get around to this, but I'm more than happy to look at a PR if you or anyone else is interested in taking a crack :)

Maybe you've already tried this, but I've heard folks mention a capacitor across Vin and Gnd on the nRF is sometimes necessary.

Markus1970, 2018-01-19:
Yes i tried already with cap up to 47uF. No chance to get it working on a wemos shield. Is there anything what i can do to get it working? Because my programming skills are wvery limited.

Thx

Markus

sidoh, 2018-01-19:
Looks like it's set to max here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Radio/PL1167_nRF24.cpp#L26

sidoh, 2018-01-19:
(oops, hit enter before i meant to).

you could change it and recompile yourself.

docs on this method here:

https://github.com/nRF24/RF24/blob/master/RF24.h#L843

Markus1970, 2018-01-19:
Lol ...thanks but i think to complie it will bi finally the problem...  ;-)

Markus1970, 2018-01-20:
Well i need to study a bit the atom editor with platformio but the amplified NRFs are now working ;-)
Many thanks at all and looking forward probably the implementation of this setting in the webif.

Markus

TehRobot, 2018-02-08:
Hey,  Having the same issue here with my "E01-ML01DP5 nRF24L01 PA LNA 2.4GHz Wireless rf Module"

I can see that line in the code in the "/master/lib/Radio/PL1167_nRF24.cpp"  But still nothing working. 

I can't even get the sniffer to work :(

Any help would be appreciated

sidoh, 2018-02-09:
@TehRobot - if listening doesn't work, it sounds more like a hardware issue.  Have you read over the [troubleshooting guide](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting)?

TehRobot, 2018-02-09:
I can access it via wifi so I know the wifi is working.  Had a similar issue with my standard nrf24 dongle until I powered it from a higher amp adaptor.  So I feel like a power issue going what ever Markus did could help me

acinsa, 2018-02-22:
Guys, I would strongly recommend a power adapter which can plug straight onto the WiFi board :
https://www.amazon.co.uk/gp/product/B00NJCB7FS/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1
These adapters take 5v from the D1/ESP8266/etc and has its own 3,3v regulator on board.

sidoh, 2018-03-13:
@TehRobot  - sorry for the late response here.

Being able to access the ESP8266 doesn't mean that the nRF is working.  I'm quite sure the ask in this ticket would have no effect on listening.  My understanding is that it only affects transmit power.

Brignoud, 2019-03-03:
> Guys, I would strongly recommend a power adapter which can plug straight onto the WiFi board :
> https://www.amazon.co.uk/gp/product/B00NJCB7FS/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1
> These adapters take 5v from the D1/ESP8266/etc and has its own 3,3v regulator on board.

Just bought a couple of these
I notice a little red led is over there.
Does this red light states for the proper function or it means a bad function error?
Anyway, it seems not a real improvement as I still notice random one of a a four or a two group spotlights (GU) not responding properly

acinsa, 2019-03-03:
Guys, here’s the combination that I used which works. It consists of the SPI breakout board with the 3.3v regulator on board. The amplifier board then plugs into this board as shown in the pics. The external aerial gives decent gain and my setup works over 50m reliably with walls in between.

https://www.amazon.co.uk/gp/product/B00NJCB7FS/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

acinsa, 2019-03-03:
attached are the pics of the SPI breakout and amplifier

Regards
Alan

https://www.amazon.co.uk/gp/product/B00NJCB7FS/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

Brignoud, 2019-03-04:
No pictures seems have been pubblished

Brignoud, 2019-03-05:
> attached are the pics of the SPI breakout and amplifier Regards Alan
> […](#)

@acinsa no pictures seems to be availables
Can you submit them again, please?

closed 2019-05-28 by sidoh