sidoh/esp8266_milight_hub #209: FUT090 support

duststorm, 2018-01-25:
I wonder, has anyone investigated the protocol the FUT090 remote (RGB+CCT with 99 channels) uses?
Would be interesting if this project would support it.

More info on the remote: http://futlight.com/productdetails.aspx?id=225&typeid=143

sidoh, 2018-01-26:
Currently there is not support for this protocol.  It's been a while since I looked, but I'm pretty sure it uses different channels and syncwords.  We'd need to sniff SPI on a remote to get these.

I'm also pretty sure there aren't all that many devices that support this protocol.  Certainly happy to look at it if folks would find it useful, and if there's someone who can help out getting the RF channels and syncwords.

sidoh, 2018-05-31:
Radio configs reversed via SPI:

https://docs.google.com/spreadsheets/d/1eMNn5HmMzNc0LgRBQlcnDYeE7KKCKBPMd15dMbIpqfM/edit#gid=0

Same as FUT092 (4 group rgb+cct) actually, except that packets are 11 bytes instead of 9 bytes.  Not getting picked up right now, but might just be some cruddy code in the nRF handler.

human101010, 2020-01-29:
I have some of these remotes and the compatible lights (the moving RGBW ones). It would be super useful to control them from an esp8266.

Any chance of support being added? I'm very happy to help with testing, sniffing commands, etc.