sidoh/esp8266_milight_hub #210: add label option to a device ID

FireyProtons, 2018-01-30:
First, awesome job here, this works amazing! Thank you so much for the ability to be able to utilize my Milight bulbs!
I'd like to be able to apply a text label to my device IDs instead of the hex value. I would prefer this for my MQTT implementation. I can try to help out... if i can in anyway. Thanks again.

RvonMallesch, 2018-02-03:
Seconded :-)

sidoh, 2018-03-13:
I like the idea of labeling stuff in the UI.  

Pushing that into the MQTT and REST API introduces a bit of complexity and overhead that I'm more cautious about.  Is this the larger part of the suggestion?

FireyProtons, 2018-03-14:
The MQTT/API would be a nice to have. The UI label feature alone would be great all by itself. Thanks!

chris9012, 2019-03-16:
Any thoughts on adding the UI label?  Would like to be able to work with my lights in the UI without having to know the hex value.  I was thinking instead of the selectize input being the DeviceID, a selectize can be the label of the device id.  That way it is more user friendly.  A separate input box would show the Device ID.  It may help though to also have a quick reference based on what is stored in settings as well for those trying to implement MQTT.  They can use the quick reference to write out their MQTT topics by linking their decription with the device ID.

poudenes, 2019-05-19:
Giving group numbers a name will make using of the Plugin easier.
When you already give the groups a name then in Domoticz you don't have rename them all.

Strontvlieg, 2019-06-15:
Yeah also group numbers with names should be great

poudenes, 2019-07-21:
Nice !! Thanks

closed 2019-08-30 by sidoh