sidoh/esp8266_milight_hub #211: Hub Led

rjsachse, 2018-02-07:
Just wondering if bridge-led type could be added?

Would be great to make a RGBWW Lamp milight gateway, also could be used as status leds.

Maybe introduce fastled libary?

sidoh, 2018-03-11:
@rjsachse are you referring to the milight hub that has a builtin LED?

rjsachse, 2018-03-17:
Yes. Like this https://encrypted-tbn3.gstatic.com/images?q=tbn:ANd9GcRGrKwZBDqOzU7BWMW0bzmP04xepuLGKTNP6YdsC6UUSgr8mZpGjEAAGnFCbg

I have done a few fastled arduino projects in the past but can't get my head around the platformio code, or how to start to implement it.

But yes would think it would be great to add hub led. I see home assistant supports hub led.

Will help as much as i can as I'm slowly learning the layout.

P.s been looking at the status indication feature and i think just release it and see what other people think about solid, blink, heatbeat feature.

rjsachse, 2018-03-22:
Ok i have been playing around the code a bit  and have got ws2802 led's and rgb leds working using FastLED now can you point me into the right direction for creating a HubLight/Bridge-Led

How would i go about reading the received packets

What do you think the best approach to implement it? Eg as a group (bridge-led,1,2,3,4,all)

sidoh, 2018-05-31:
Oh -- I was misunderstanding you in the other ticket, @rjsachse.  Now I get what you were suggesting.

sidoh, 2018-05-31:
I think this project is not well suited for this kind of thing.  I would suggest @bruhautomation's [ESP8266 LED controller](https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs) instead.

closed 2018-05-31 by sidoh