sidoh/esp8266_milight_hub #212: Does espmh listen to responses from v6 (RGB+CCT) bulbs?

MisterC925, 2018-02-08:
I took a look at the code and I didn't see it but I'm still coming up to speed.

I've noticed a few of my RGB+CCT bulbs (the "v6" protocol) get missed every once in a while (even with 150+ repeats).

From what I've read, these bulbs send some form of ACK when lit?   Does the EspMH listen for that ack or does it simply send the command X number of times?

Thanks!

sidoh, 2018-02-09:
The only bulb-originated traffic I'm aware of is detailed in [this ticket](https://github.com/sidoh/esp8266_milight_hub/issues/91).  It's basically a mechanism bulbs use to synchronize themselves while in scene mode.

What I can say is ACKs are probably not on the same channels or use a different syncword.  No traffic on the same configs from the bulbs except for that mentioned in #91.

I would be very excited to learn that the bulbs send an ACK when they receive a command, and would definitely implement it. If you have any additional information, please do share :)

MisterC925, 2018-02-09:
I think I assumed it due to the fact that the V6 UDP bridges have some kind of two-way communication, and since the gateway is (or at least was up until v6) a simple wifi to serial converter, I thought maybe the response from the v6 protocol was from the bulb itself.   It is entirely possible it's just from the gateway instead.

Unfortunately, as you know, Futlight seems to like to keep things closed.

Thanks

sidoh, 2018-02-09:
Gotcha.  Unfortunately, the V6 UDP protocol is vastly different than what's sent over air.

MisterC925, 2018-02-09:
Okay I'll close this for now but if I find any more info I'll open it.  Even with the occasional misses, the ESPMH is 1000x better than the old way I used to do it (HA-Bridge + Wifi box)  so thank you for all of your work on this.

closed 2018-02-09 by MisterC925