sidoh/esp8266_milight_hub #234: Special character '+' not supported in mqtt password

diabl0w, 2018-02-18:
Commands sent through mqtt don't work if a '+' is present in the mqtt password. Oddly enough, my broker says that the esp8266 is subscribing successfully in logs when first started up, but commands never go through. Without changing any configuration besides the password, commands now work.

sidoh, 2018-02-18:
Aah, that's probably the fault of the crappy form handling in the UI.  Thanks for reporting.

If you're in a pinch, you can probably set it with the API.  I doubt this is happening at any level below the UI.

diabl0w, 2018-02-18:
Thanks for the quick reply, at the moment it's easier for me just to change
the password. If I ever decide to try the API and it is still not working
then I'll report back.

sidoh, 2018-03-12:
I think I'm not able to reproduce this.  Seems to work fine in my environment.

May I ask what browser you're using, @diabl0w?

diabl0w, 2018-03-13:
I know for a fact it happens on Android mobile using Chrome, I just double
checked. I will try on Firefox on my desktop when I get a chance.

diabl0w, 2018-03-13:
Yes it does occur on Firefox as well on desktop. Note that you must hit the submit button and then refresh the page to see the effect. Upon refreshing, the '+' symbol is replaced by a space

sidoh, 2018-03-13:
Interesting.  I was not able to reproduce it because it's fixed in 1.7!  I reverted back to 1.6 and it happened consistently.

The error was on the server end, not the client.  The client is sending the correct data.

One of the changes in 1.7 (thanks to @cmidgley!) is using the upstream HTTP server instead of the locally copy/pasted version, which included fixes for bugs which have since been patched.  

I think the upstream HTTP server must've patched this bug as well.

We have @cmidgley to thank for this one! :D

diabl0w, 2018-03-26:
great news! I'll wait for the release to confirm... thanks

closed 2018-05-05 by sidoh