sidoh/esp8266_milight_hub #237: Brightness sync problem

badstraw360, 2018-02-26:
Ok I'm still having the same problem... I haven't been able to open issue [https://github.com/sidoh/esp8266_milight_hub/issues/191](url)

When I set the brightness to 40% and then close the browser, wait a few seconds and open it again, the brightness slider will reset to 100% but the bulb physically stays at 40%. So if I want to fix that I need to manually move the slider to update the bulbs again.

I'm using hassbian 0.64.0

That makes HomeKit to get wrong data, the bulbs in HomeKit always update to 100% no matter what. 

I would appreciate your help. :)

```2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0xF63/rgb_cct/1
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0xF63/rgb_cct/2
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x5555/rgb_cct/2
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0xF63/rgb_cct/4
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x5555/rgb_cct/1
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x1111/rgb_cct/0
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x0/rgb_cct/1
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x2222/rgb_cct/0
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to lights/states/0x9999/rgb_cct/0
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to /blind/bc/state
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/1: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x5555/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/4: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x5555/rgb_cct/1: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x1111/rgb_cct/0: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x0/rgb_cct/1: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x2222/rgb_cct/0: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x9999/rgb_cct/0: b'{"state":"OFF","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:54:43 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on stat/sonoff/POWER: b'OFF'
2018-02-26 09:55:28 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:55:29 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:03 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:06 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":48,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:08 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":25,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:19 ERROR (MainThread) [homeassistant.core] Timer got out of sync. Resetting
2018-02-26 09:58:24 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":80,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:25 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-02-26 09:58:26 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
```

badstraw360, 2018-03-05:
Ok, just set the brightness to 50% in HA and then close your browser and open it up again. The brightness slider will reset to 100% but the bulb physically stays at 50%.

I installed hassbian again, and it keeps doing that.

I would really appreciate your help.
I can record a video of the issue if you need, or let me know how I can help you to diagnose that issue. 
Thanks again.

sidoh, 2018-03-06:
Are you able to try Mosquitto?  There's at least one behavior I've seen out of HASS' integrated MQTT broker that makes me think it's not the smartest / most robust implementation in the world.

badstraw360, 2018-03-06:
Sure... I'm using mosquitto right now with the lights, just let me know how I can help.

sidoh, 2018-03-11:
@badstraw360 I was being a dolt and totally misunderstood your issue.  I thought this was behavior you were seeing when HASS restarted.

I am not able to reproduce this.  If I set brightness to 50%, the slider stays at 50%, even if I close the tab, wait for a few minutes, and open it again.  It's also consistent across devices.

It sounds like some automated thing is causing the brightness to get set to 100% after you set it to 50%.  Can you think of any reason that'd be happening?  Is the timing in your HASS logs consistent with that?

badstraw360, 2018-03-11:
Actually Im using hassbian, maybe HASSIO is working fine... 

I tried again turned on the bulb and set brightness to 12%, then click on refresh page and the slider went back up to 100% but the bulb stayed at 10%, then set the slider to 95% and the light went back up to 95%. 

```2018-03-11 20:22:36 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0x9999/rgb_cct/0: b'{"state":"OFF","color_temp":294,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-03-11 20:24:29 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-03-11 20:24:34 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":12,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-03-11 20:24:46 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":96,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-03-11 20:24:48 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"ON","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
2018-03-11 20:24:50 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on lights/states/0xF63/rgb_cct/2: b'{"state":"OFF","level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}'
```

It's super weird, I turned off all my automations and it's still doing it. If I recall correctly it was doing it since I was using HASSIO, I had no automations, groups or scripts back then, It was a brand new installation.

sidoh, 2018-03-11:
Just to make sure I'm following, it does look like you're getting MQTT messages consistent with this behavior, correct?

If that's the case --

I'd try to find where the messages are coming from.  Perhaps try enabling debug logging on mosquitto?

If they're coming from espMH, can you include a dump of your settings (just go to `/settings`)?  Make sure to censor passwords.  Might also need to get serial logs with some debugging enabled.

badstraw360, 2018-03-11:
Yes Im getting MQTT messages consistent with this behavior...

I enabed logging on mosquito: 

[mosquitto.log](https://github.com/sidoh/esp8266_milight_hub/files/1800902/mosquitto.log)

My settings:

`{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.xx:1883","mqtt_username":"xxxx","mqtt_password":"xxxx","mqtt_topic_pattern":"lights/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"lights/states/:device_id/:device_type/:group_id","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"device_ids":[0,4369,8738,13107,21845,39321,3939],"group_state_fields":["state","level","hue","saturation","mode","color_temp","bulb_mode","computed_color"]}`

I really appreciate your help..!

sidoh, 2018-03-11:
Don't see anything obvious in the mosquitto logs.  Appears that espMH is the only thing sending state updates.

Would need Serial logs from firmware compiled with debug flags.  `DEBUG_PRINTF` and `MQTT_DEBUG` should be sufficient.  You can enable those here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/platformio.ini#L28

(I monitor serial with `pio device monitor` while the ESP8266 is plugged into my laptop)

badstraw360, 2018-03-11:
I just uncomment these lines and upload the firmware again, right?

`# -D DEBUG_PRINTF
# -D MQTT_DEBUG`

to 

` -D DEBUG_PRINTF
  -D MQTT_DEBUG`

Where are the logs stored?

sidoh, 2018-03-11:
Yes, although make sure they're on the same line with `build_flags`.

They're serial logs.  If you have platformio installed, just use the command `pio device monitor`.

badstraw360, 2018-03-11:
This is what I'm getting: 

`MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE A5 63 31
Elapsed: 142
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE B2 63 32
Elapsed: 142
Sending packet (50 repeats):

00 DB E4 42 61 3A B3 63 AA
Elapsed: 141
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE B0 63 40
Elapsed: 143
Sending packet (50 repeats):
00 DB E4 42 61 6B B1 63 45
Elapsed: 140
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE AE 63 3E
Elapsed: 142
Sending packet (50 repeats):
00 DB E4 42 61 5F AF 63 37
Elapsed: 142
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CB AC 63 25
Elapsed: 142
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE AD 63 39
Elapsed: 142
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2

MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE 9A 63 3A
Elapsed: 141
Sending packet (50 repeats):
00 DB E4 42 61 49 9B 63 1D
Elapsed: 141
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE 98 63 28
Elapsed: 142
Sending packet (50 repeats):
00 DB E4 42 61 6B 99 63 4D
Elapsed: 142
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CE 96 63 26
Elapsed: 141
Sending packet (50 repeats):
00 DB E4 42 61 5F 97 63 3F
Elapsed: 142
MqttClient - publishing update to lights/states/0xF63/rgb_cct/2
MqttClient - Got message on topic: lights/commands/0xF63/rgb_cct/2
MqttClient - device 0F63, group 2
Sending packet (50 repeats):
00 DB E4 42 66 CB 94 63 2D
Elapsed: 141`

sidoh, 2018-03-12:
Everything I'm seeing is indicating that espMH is behaving normally.  The state updates it's sending out appear to be due to commands it was sent.  This is consistent with the mosquitto logs as well.

It seems like something is causing HASS to reset the brightness to 100%.

badstraw360, 2018-03-12:
So would you recommend to reinstall HA? I don't really know what to do at this point. :/

sidoh, 2018-03-12:
I would expect stuff to show up in the HASS log.  If not, enable DEBUG logging for all components.

badstraw360, 2018-03-12:
I think I've found the problem!!! I had this option disabled:

[image]

Thank you very much for your work!! I'm really glad.. 

Just one quick question... Can I use "transition" to increase the brightness gradually?

sidoh, 2018-03-12:
Aah.  I should've caught that.  You can probably disable "level".

No support for transitions from espMH currently (I'd like to add support in the future), but I believe HASS can drive transitions.

Glad you got it working!

closed 2018-03-12 by sidoh