sidoh/esp8266_milight_hub #238: Gateway Server v6 not working

RudolfRendier, 2018-03-02:
I'm having trouble setting up the esp8266_milight hub as drop-in replacement for my Milight Wifi iBox (v6) that was controlled by HomeAssistant.

I put together the hardware, flashed the latest firmware, connected it to wifi and paired my rgbww-bulbs to a fresh device_id. I can control the bulbs from the web-ui just fine with rgb+cct

I configured a gateway server for the new device_id, listening on the default v6 port: 5987
And configured it for protocol version 6.

In the logs of HA I can see it's trying to control the bulbs (setting up pipelines i.e.) but the bulbs are not responding.

```json
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"gateway_configs":[[52614,5987,6]],"group_state_fields":["state","brightness","computed_color","mode","color_temp","bulb_mode"]}
```

In my Home Assistant-config I modified as little as possible; I remove the bridge-led:

```yaml
- platform: limitlessled
  bridges:
    - host: 192.168.1.177
      version: 6
      groups:
      - number: 1
        type: rgbww
        name: slaapkamer
        fade: on
```

```yaml
- name: slaapkamerFel
  entities:
    light.slaapkamer:
      state: on
      rgb_color: [231, 208, 189]
      brightness: 100

- name: slaapkamerNormaal
  entities:
    light.slaapkamer:
      state: on
      rgb_color: [231, 208, 189]
      brightness: 76

- name: slaapkamerKodiPlaying
  entities:
    light.slaapkamer:
      state: on
      rgb_color: [231, 208, 189]
      brightness: 20

- name: slaapkamerUit
  entities:
    light.slaapkamer:
      state: off
```

I know that mqtt is probably superior, but I was hoping this would just work.
But it doesn't and maybe I've misconfigured something or maybe there's a bug lurking in the shadows.
What can I do to (help) get this resolved?

bombcheck, 2018-03-03:
Can you set a port for the bridge in the HA config? Maybe try a different port (will be neccessary when you need to config two or more bridges).

RudolfRendier, 2018-03-03:
Thanks, that did the trick.

Configured gateway server of milight_hub with port 6000 and HA as well and it started working.
Explicitly using the default port now works as well.

Reverting back to my previous config, without explicit port in HA also works.

Turns out to be not an issue at all.

closed 2018-03-03 by RudolfRendier