sidoh/esp8266_milight_hub #239: MQTT Not working after Home Assistant Restart unless MiLight hub used.

TehRobot, 2018-03-05:
I have setup and are using MiLight bulbs through MQTT however every time I restart Home assistant I am unable to use the bulbs again until they are turned off and on via the Milight Hub webpage.

Just wondering if there is a fix for this or something I may have done wrong.

I am using 
Home Assistant 64.3,  
MiLight version 1.6.5,
HASS default MQTT
NodeMCU

sidoh, 2018-03-05:
I also use MQTT with HomeAssistant, and I have not encountered this.  Could you paste your HASS config and the output of the `/settings` route for espMH (make sure to censor passwords)?

TehRobot, 2018-03-05:
Configuration.yaml
`
mqtt:

light:
 - platform: mqtt_json
   name: "Dining"
   state_topic: "milight/state/0x1928/rgb_cct/2"
   command_topic: "milight/command/0x1928/rgb_cct/2"
   brightness: true
   rgb: true
   color_temp: true
   optimistic: false
   qos: 0
`

Milight Settings output
`
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"radio_interface_type":"nRF24","packet_repeats":150,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"10.1.1.251","mqtt_username":"homeassistant","mqtt_password":"**********","mqtt_topic_pattern":"milight/command/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/update/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/state/:device_id/:device_type/:group_id","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"device_ids":[4096,6440,1649],"gateway_configs":[[6440,48899,6],[1649,48899,6],[4096,48899,6]],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect"]}`

sidoh, 2018-03-07:
It appears you're using HASS' embedded MQTT broker.  Only thing I can suggest offhand is to try Mosquitto.  I've seen a few things that make me believe HASS' broker doesn't follow specs super closely.

It might also be helpful to see what the retained state messages look like, and if they match your expectations.  That would be where HASS is getting its initial state from.

I can try reproducing with my HASS instance, but I've done this before and have not encountered it.

TehRobot, 2018-03-07:
I have fixed it. 

I installed mosquito and had the same issues (however could use mosquito_sub which was handy).

In the end I realised it was the customise I had put in for the lights when using the original MiLight hubs as the update in 64.3 had turned them to lighting bolts instead of toggle switches "assumed_state: false" was needed to retain the toggle switch.  Once I removed this it is working normally and has no issues. 

Thanks for the help and time. Cheers

closed 2018-03-07 by TehRobot

sidoh, 2018-03-07:
Glad you got it working!  Thanks for following up with the fix.