sidoh/esp8266_milight_hub #244: Is the Beat PlatformIO still required?

cmidgley, 2018-03-11:
In the README, it says: 

> Note that currently you'll need to use the beta version of PlatformIO. To install with pip:
> pip install -U https://github.com/platformio/platformio-core/archive/develop.zip

I found that I was successful with the standard/production version of PlatformIO.  Perhaps we should remove this from the README?

sidoh, 2018-03-11:
Yeah, looks like they released a version that had the needed feature.

Actually looks like the esp8266/Arduino folks released 2.4, which had the feature (KRACK bugfix) that I needed pio beta for anyway.

Great catch!

jstans, 2018-04-30:
The milight gateways being susceptible to KRACK is part of the reason I ended up here and haven't set up an isolated subnet. Out of interest, is the comment you made in the README.md regarding 2.4ghz being insecure related to this @sidoh ? (and thus no longer relevant?)

sidoh, 2018-04-30:
The comment in the README is referring to the 2.4 GHz communications sent to the bulbs, not Wifi.  The packets are completely unsecured and trivial to spoof.

PR #245 (2018-03-12): Remove stale instructions for PlatformIO; don't use staging version of Arduino SDK
@cmidgley this fixes #244
closed 2018-05-05 by sidoh