sidoh/esp8266_milight_hub #248: Is my light supported?

olskar, 2018-03-17:
Hello!

I am trying to pair my bulb with the milight hub but I can not get it to work.

Earlier I used Rflink that was able to pair the light by sending the command:
10;MiLightv1;20F746;01;34BC;PAIR;

So I know both the bulb and sender is working fine.

Is there a way to see if my bulb is supported or not? It is a RGB+CCT - aka RGBWW

rjsachse, 2018-03-17:
Looks like yes your globe is supported. Have you used the sniffer or try to pair a bulb to say 0x123A

rjsachse, 2018-03-17:
Have you read https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs

olskar, 2018-03-17:
Thanks, redid everything with another esp8266 :) now works!

closed 2018-03-17 by olskar

sidoh, 2018-03-17:
Thanks, @rjsachse!

Glad you got it working, @olskar.