sidoh/esp8266_milight_hub #250: grouped lights won't change color at the same time

badstraw360, 2018-03-18:
If I combine multiple lights into one entity to make group and run an automation, the lights turn on and change color one after the other. Is it possible to make the lights turn on and change color at the same time like one single light? I tried to make a group, and I even tried to use the new "light group" component but the outcome is the same, the lights change color one by one. 

```
light:
  - platform: group
      name: downstairs
      entities:
        - light.pasillo
        - light.sala
        - light.comedor
```
```
scenes:
  - name: party
    entities:
      light.downstairs:
        state: on
        rgb_color: [140,90,255]
        brightness_pct: 80
```

```
automation:
  - alias: "SONOS play"
      trigger:
        - platform: state
          entity_id: media_player.sonos
          from: 'idle'
          to: 'playing'
      action:
        service: scene.turn_on
        entity_id: scene.party

```
I'm not sure if that is a problem with my setup, or if it's a limitation of the protocol.  

Thanks for this great piece of software, it's really awesome. :)

sidoh, 2018-03-18:
Unfortunately this is a limitation of the protocol.  If I'm understanding your config correctly, it has to send three separate commands for any change to the `downstairs` group.

Some bulbs support being paired to more than one device id / group id.  A workaround might be to pair all three groups of bulbs with a common device/group.  That way you can address all of them at the same time with a single command.

badstraw360, 2018-03-20:
Actually, it has to send one single command to a group of lights. 

I have 3 bulbs paired to one device Id / group id, then I have 2 additional bulbs paired to another device id/ group id, then 3 additional bulbs paired to a different device id / group id. So I have 8 bulbs in total. 

Im gonna try to see if I can pair a few bulbs to the same group Id, maybe that would work... like you're saying. 

Thank you, I'll let you know how it goes.

sidoh, 2018-03-31:
@badstraw360, did you get this working?

badstraw360, 2018-04-02:
Actually I paired pair all three groups of bulbs with a common device/group, and they all work like a single bulb (if I'm using milight hub). But the main issue is that HA won't detect state changes, so for example If I turn off the lights I'm unable to turn them back on. 
It's weird because if I create a group and send a command to turn them on, the lights turn on one by one, but after that, I can't do anything because the lights won't respond to any command. 

So I don't know if that happens because HA can't control several lights at the same time or what.. 

Thank you for spending some of your time to help me with this, I appreciate it…

sidoh, 2018-07-22:
Hey @badstraw360, sorry that this slipped.

Correct me if I'm wrong but it sounds like what you're trying to do here is assign bulbs to groups so they can be controlled with a single command.  In addition, sounds like you want to assign them a different ID so that they can be controlled individually.

Is this right?

espMH isn't going to do a good job of untangling state in this scenario.  You'd have to have logic on the HASS end to deal with that.

sidoh, 2019-05-19:
Hey @badstraw360, sorry this slipped for nearly a year!  Working through the issue backlog.

Sounds like we came to a resolution, so going to close.  But please let me know if you'd like to discuss further.

closed 2019-05-19 by sidoh