sidoh/esp8266_milight_hub #253: Option to restart the milight hub at set intervals or via MQTT request

MimbaMonkeyHouse, 2018-03-30:
Several times a day my ESPMH devices seem to go unresponsive to requests. This seems to happen much more frequently since I set scenes on Home Assistant triggering multiple requests to multiple bulbs at the same time. 
When this happens, the ESPMH still appears on the network. I can log on and restart the device from the interface, which then functions as normal again for a while. I'd like to automate this restarting process to avoid having to log on to each of my 3 gateways multiple times per day.
Is there any way to have the devices restart at set intervals or to send an MQTT or http request to restart the hubs? If not, is this a feature that can be implemented easily?

sidoh, 2018-03-30:
I've noticed this sort of behavior when the TCP stack on the ESP gets overwhelmed.  Think it throttles packets and things get screwy.

Ideally this wouldn't happen, but unfortunately there's not an easy fix.  Probably the right approach would be to use the RTOS SDK, but that'd be a huge rewrite.  Don't even think there's an Arduino version of the RTOS SDK.  Some other less major things that might help I have in mind, but not sure when I'll have time to get to them.

You might also try the suggestions on the [Performance Tuning](https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning) page.  Repeat throttling in particular might help in your case.

Now to answer your actual question :)

You can already restart via HTTP.  The UI is just a static page that calls the REST API.  Example:

```
curl 'http://milight-hub.local/system' -X POST -H 'Content-Type: application/json' -d '{"command":"restart"}'
```

Not currently possible to do this with MQTT, but hopefully this suits your needs.

MimbaMonkeyHouse, 2018-03-30:
Hey @sidoh,

Thanks for this quick and complete answer. I did look at the Performance Tuning tips and still get this behaviour.
For now I think I'll just set a Command Line automation via Home Assistant using the REST API. This is exactly what I needed and I am keen to try it as soon as I get home.
Cheers for your hard work!

sidoh, 2018-03-31:
Glad that works for you!

closed 2018-03-31 by sidoh

sidoh, 2018-04-02:
> Can I just replace 'http://milight-hub.local/system' with the address of my hub? No need for username and password?

Yes, that's right.

> The automation is triggered, no error in the log, but I'm not sure how to find out whether the ESPs get restarted or not...

I'd leave a ping running and trigger the automation manually.  If the ping stops running it's a good sign the reboot took place.

MimbaMonkeyHouse, 2018-04-02:
OK, great. I'll figure out how to leave the ping running... Thanks for the help!