sidoh/esp8266_milight_hub #254: FUT 089 (8 group remote): Group 7 and group 2 interferences?

devax, 2018-03-31:
Hey guys, this is a very weird issue.
I'm using a FUT 089 (8 group remote) in combination with ESP8266 Milight Hub. Milight Hub is controlled by Openhab through MQTT.
So here is what I observe:

1. Bind a RGB_CCT lamp to group 2
2. Bind 4 RGB_CCT lamps to group 7
3. Switch both lamps ON
4. Send this command on MQQT (triggered through Openhab): `{"state":"ON","level":40}` on topic `milight/commands/0x66CA/rgb_cct/7`
5. On MQTT, Milight Hub will correctly publish the "state" of goup 7, as expected.
6. Observe lamp on group 2 will turn OFF (totally unexpected).

When I find the very same lamp to group 3 instead of group 2, everything works normally. Therefore I suspect it is a bug in ESP Milight Hub.

I have tried this with current master (1.6.5) and also with the just released 1.7.0-dev4. Same behavior with both versions.

devax, 2018-03-31:
On group 2 there is a FUT013, on group 7 there are four FUT104 (spotlights).

sidoh, 2018-03-31:
Do you have a real remote to test with?  Sort of suspect your devices just half-support FUT089.  The state wouldn't be decoded correctly if there were a collision in the RF packets being sent.

devax, 2018-04-01:
I have a real FUT089 device here and using the device it works well. I paired the lamps using the device and use the devices ID as part of the MQTT topic to transfer commands.

sidoh, 2018-04-01:
Could you provide samples of sniffed packets?  I have one as well, and I'm pretty sure the protocol is emulated correctly.  I'll double-check later this weekend.

sidoh, 2018-04-01:
I included some packet captures from both espMH and a B7 wall remote below.  As far as I can tell, the protocol is mirrored perfectly.

Crucially, `command`, `argument`, and `group` fields all match up.

-----

### espMH, group 7 off:

```
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 23 66 C0 A7 60 1D 

Decoded:
Key      : 00
b1       : 25
ID       : 0000
Command  : 01
Argument : 10
Sequence : 15
Group    : 07
Checksum : 0A
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 23 66 C0 A6 60 20 
```

### espMH, group 2 off:

```
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 23 66 D7 B1 63 11 

Decoded:
Key      : 00
b1       : 25
ID       : 0000
Command  : 01
Argument : 0B
Sequence : 1B
Group    : 02
Checksum : 06
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 23 66 D7 B0 63 14 
```

### B7 wall remote, group 7 off:

```
fut089 packet received (9 bytes):
Raw packet: 82 25 35 14 52 14 91 10 D7 

Decoded:
Key      : 82
b1       : 25
ID       : 66BE
Command  : 01
Argument : 10
Sequence : 14
Group    : 07
Checksum : AB
fut089 packet received (9 bytes):
Raw packet: 33 BC 34 26 37 26 98 49 62 
```

### B7 wall remote, group 2 off:

```
fut089 packet received (9 bytes):
Raw packet: 68 70 93 1D BE 1F 1D BB D1 

Decoded:
Key      : 68
b1       : 25
ID       : 66BE
Command  : 01
Argument : 0B
Sequence : 17
Group    : 02
Checksum : 7E
fut089 packet received (9 bytes):
Raw packet: EB 04 FC 5E 7F 57 DD 96 D5 
```

devax, 2018-04-01:
Sorry for the delay! 
On MQTT, this is what gets published: 
`milight/commands/0x66CA/rgb_cct/7` - `{"state":"ON","level":57}` This gets produced and published by the Openhab binding.
Because ESP Mylight Hub will not act on that command on that topic (the binding cannot publish to "fut089", but only to "rgb_cct"), I have a MQTT bridge in place which will forward the very same message to this topic: `milight/commands/0x66CA/fut089/7`. 
These are the packets I see with the sniffer:
### Packet 1 
fut089 packet received (9 bytes):
Raw packet: 00 D8 FB E9 62 A9 3C 60 B7 

Decoded:
Key      : 00
b1       : 25
ID       : 66CA
Command  : 05
Argument : 39
Sequence : 8E
Group    : 07
Checksum : E0
### Packet 2
fut089 packet received (9 bytes):
Raw packet: 00 D8 FB E9 66 CB 3F 60 80 

Decoded:
Key      : 00
b1       : 25
ID       : 66CA
Command  : 01
Argument : 07
Sequence : 8D
Group    : 07
Checksum : A9
### Packet 3
rgb_cct packet received (9 bytes):
Raw packet: 00 DB FB E9 61 98 ED 60 ED 

Decoded:
Key      : 00
b1       : 20
ID       : 66CA
Command  : 04
Argument : C8
Sequence : 5F
Group    : 07
Checksum : 3A
### Packet 4
rgb_cct packet received (9 bytes):
Raw packet: 00 DB FB E9 66 CB EC 60 24 

Decoded:
Key      : 00
b1       : 20
ID       : 66CA
Command  : 01
Argument : 07
Sequence : 5E
Group    : 07
Checksum : 75

After that, the light on group 2 will be turned off.
Looking at the packages, I agree, this does not look like an ESP Mylight Hub issue. Probably there is something wrong with the OH binding and/or my rules. I have excluded that possibility first by disabling all rules and even the thing/item for the lamp on group 2 and still have seen the effect. I am still confused.

sidoh, 2018-04-01:
Is the openhab binding using UDP?  There's a MQTT binding that might work better:

https://community.openhab.org/t/use-esp8266-to-control-unlimited-groups-of-milight-or-limitless-led-bulbs-in-openhab2/34906

devax, 2018-04-01:
This is the binding I'm using.

sidoh, 2018-04-01:
If it seems like the issue is with the binding, I'd recommend dropping in on that post.  The author has been very responsive in the past.

It's possible that the binding was authored with the 4-group protocols in mind, and just needs to be adapted for the 8-group protocol.

@Skinah is the author of the MQTT binding.  Perhaps he could help out on this one?

Skinah, 2018-04-02:
@sidoh 
Thanks for the heads up this post was here, I'll make some changes to the binding to bring it inline with what the hub does. 

I'm guessing this causes an issue because it is group 7 with remote type rgb_cct
milight/commands/0x66CA/rgb_cct/7

I'll add support so the binding sends
milight/commands/0x66CA/fut089/7

@devax
Try the new binding out first and if it does not fix, please test sending the MQTT manually with mosquitto_pub command in linux or another application to rule my binding out as a cause. Also I have found rarely that the bulbs will sync to multiple remotes/groups if someone is pushing the ON button on a remote whilst they are turning to power on. You have to train your family about this.

From your description I suspect it is more likely to be an issue due to using group 7 for a rgb_cct which I agree my binding should not do (and wont in next version), but the hub should gracefully handle it by ignoring groups above 4 for rgb_cct.

Do some testing and then let sidoh know if the hub was not liking group 7 with the wrong remote type and he can then decide if that is an issue or not.

devax, 2018-04-02:
Hey guys, thank you very much for your effort putting in this issue!
Most important: With the new binding, the issue is resolved. This is because commands now are sent directly to `milight/commands/0x66CA/fut089/7` and the hub will handle that well. So all is good for me, now.
Nevertheless, this is a fact:
When I publish the message `{"state":"ON"}` to the topic `milight/commands/0x66CA/rgb_cct/7`, my light that is bound to group 2 of the fut089 remote with id 0x66CA turns off! This is without going through Openhab/the binding, this is only MQTT communicating with the hub (using 1.7.0-dev4 atm). I even stopped my OH server to make sure it is not some reaction from OH to the states update or anything. So it should be reproducable by other ppl having a fut089 and a light on group 2 and 7.
Well, under normal circumstances no one should publish to the topic .../rgb_cct/7, you should use .../fut089/7 instead, but old OH binding was publishing to that topic so that caused the switching off of lamp on group 2.
Not sure if it is worth fixing in the hub.

sidoh, 2018-04-02:
Oh, I bet what was happening is trying to cram group 7 into the RGB+CCT packet structure.

Really we just need some good ol' fashioned input validation here :)

sidoh, 2018-04-02:
Going to close since it seems like your problem was resolved (feel free to reopen if not), but I'll open a separate issue to validate input

closed 2018-04-02 by sidoh

devax, 2018-04-03:
Awesome, thank you!