sidoh/esp8266_milight_hub #255: FUT089 saturation changes color_temp in white mode via control panel controls

Skinah, 2018-04-02:
If you use the controls in the hubs control panel, the ones found using the IP address in a browser the saturation control will send MQTT states that have the color_temp change as you move the saturation control. Controls work fine in color mode, this is only when the globe is in white mode.

milight/states/0x99/fut089/7 {"state":"ON","color_temp":153,"bulb_mode":"white"}

sidoh, 2018-04-02:
Try selecting a hue.  Color mode is only set when the hue is known.

sidoh, 2018-04-02:
Oh, misunderstood.  But the reason is related.  The FUT089 protocol uses the same command ID for both color temp and saturation settings (no idea why).  So if the bulb mode is white, it'll send the right packet, but interpret it incorrectly later.

I suppose the options are:

1. Refuse to set saturation when bulb mode is not color (this seems unnecessarily confusing)
1. Force the bulb into color mode by setting a default hue (or last known hue) if one is not provided

sidoh, 2018-04-28:
I think the automatic mode switching added in 1.7 kind of already accomplishes this.  It's a little confusing because it doesn't permanently switch into the relevant mode, but I think it at least solves the problem.

closed 2018-05-05 by sidoh