sidoh/esp8266_milight_hub #263: Feature request? Set ch/freq to set value by user.

Skinah, 2018-04-24:
Hi Sidoh and thanks for the work being done on this,

I have now built two hubs with the goal of getting the globes to change states faster as I have 23 globes running great here but it takes around 2 seconds for all of them to turn ON. From reading another issue thread where you replied I believe the channel/frequency is fixed and not scanning?
What I would like to do is set one hub to broadcast on say Channel 1 and the second hub on Channel 3 and not scanning (fixed). This way as the globes scan through the 6 channels they pick up the RF packet faster from whichever hub they hear first. Does this make sense as to why and what I am asking be added as a feature?

sidoh, 2018-04-25:
Definitely makes sense why you'd want this.  I would like to add more flexibility to how channel scans are handled.  

Might be other places where @khamann recommended stuff like this, but I think [this comment](https://github.com/sidoh/esp8266_milight_hub/issues/184#issuecomment-352610491) covers it pretty well.

Right now every packet is sent on 3 channels, and 3 channels for every protocol are scanned.  I think the idea would be to expose a setting to control this behavior.  Perhaps could just let the user select any combination of hi/med/low channels for both send and receive.

sidoh, 2019-03-17:
Added the ability to select which channels sends go out on in #421.  This doesn't affect which channels are listened to, though.  Actually only the LOW channel is listened on.  Will add a setting for that as well.

Skinah, 2019-03-17:
Great news and thanks for the work done on this.

closed 2019-05-28 by sidoh