sidoh/esp8266_milight_hub #266: Newbie sniffing question

xbmcnut, 2018-04-28:
Great work so thank you for this. I intend using it to replace a dead v4 Milight bridge and with Hass.io

I have flashed 1.6.5 onto a NodeMCU V2 and have a NRF24L01 module (that was working with My Sensors previously) and I have managed to pair one MiLight bulb (CCT) and can control on/off, brightness and color temp. Based on this, I'm assuming the NRF module is fully working?

My question is I have two different 2.4Ghz RF remotes; one for a flash pendant wave light and another for some RGB LED strips. Both lights work with their consecutive remotes.

When I activate sniffing though and use either remote, I see nothing. I do see data when I use the UI to control the one paired bulb but nothing when using the remotes. Am I supposed to see something when the remote turns on my lights that are not paired with any bridge? 

P.S I have not configured anything in the MQTT section yet other than my broker address.

sidoh, 2018-04-28:
Any traffic espMH intercepts while in listening mode should show up.  Shouldn't matter if the remotes are paired with anything.

If it's not showing up, it might be a remote model that's not supported.  Could you:

* Let me know what model the remote is?  [futlight.com](http://futlight.com/) seems to have the best product catalog.
* See if there's anything interesting in Serial logs with the `DEBUG_PRINTF` flag enabled?

xbmcnut, 2018-04-28:
Thanks for the prompt reply. The LED RGB strip remote appears to be a FUT098. The other one is not shown at futlight.com but here is picture.

[image]

I'm assuming I'll have to compile to enable the debug flag?

xbmcnut, 2018-04-30:
I there. Bit rusty on Platformio but tried it with settings below.

[image]

Serial monitor inside Platformio have me nothing but 'null null null null' so I connected to COM4 with Putty. Got valid data when I used an actual FUT007 (found the one that came with my bridge), but the remote pictured above and the FUT098 show nothing. Maybe there are just outside the band of the NRF2401 I have?? I have a LT8900 (with NRF2401+ chip) so maybe I'll try that next.

xbmcnut, 2018-04-30:
The other module I have is not a LT8900 (as sold to me) but the ultra mini version of the NRF2401+. I hooked it up anyway and get the same results from PuTTy.

```
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 10.0.1.182
Setup complete
Radio is available

                  Packet received: E05A58D540B09778EE030303Packet transformed: 075A1AAB020DE91E77C0  C0C0Failed CRC: expected 0, got 49344
```
Only see these results when pushing buttons on the 'real FUT007. Nothing from the other remotes. Does that tell us anything?

sidoh, 2018-04-30:
Sniffing on the real LT8900 might even be broken.  So maybe a good thing it wasn't a real one? :)

Definitely looks like an unsupported protocol.  Might even suggest it's knockoff hardware that wasn't implemented to spec.

Could you get a few more packet captures and label them with what button they correspond to?

xbmcnut, 2018-04-30:
Thanks but that's the thing. The knock off remotes do not generate any code on PuTTy even though the work on their respective lights.

sidoh, 2018-04-30:
Sorry, I should've been more clear.  I meant the logs from serial output.  Like the stuff you pasted above:

```
Packet received: E05A58D540B09778EE030303Packet transformed: 075A1AAB020DE91E77C0  C0C0Failed CRC: expected 0, got 49344
```

This contains the packet data.

sidoh, 2018-04-30:
If you're seeing that "Packet received" line, you successfully compiled with DEBUG_PRINTF.

If you're not seeing serial output for your remotes, then they're probably sending on different channels and with different syncwords.  We'd likely need to sniff the SPI bus to get those parameters.

I ordered a FUT007 from ebay, but won't be here for a few weeks.

xbmcnut, 2018-04-30:
Thanks for all your help, much appreciated. Is there a way I can set up a sniff from the SPI bus?

sidoh, 2018-04-30:
Doing this is quite a bit more heavyweight than most things required for this project.  You'd need a logic analyzer or oscilloscope, and you'd probably need to take apart the remote so you could attach probes to the appropriate pins.

xbmcnut, 2018-04-30:
I have an oscilloscope and an electronics background and don' like being beaten by a machine! :) If I can help get these remotes working, I'm happy to help if you can point me in the right direction.

sidoh, 2018-05-01:
It's not something I actually have any experience with.  I think people basically find the microcontroller or the radio, figure out where the SPI pins are, and watch for messages as actions are taken.  Fancier scopes can decode SPI messages.  Some of the folks who have helped out with this before have digital logic analyzers that do this.

xbmcnut, 2018-05-14:
Thanks, I appreciate you going to that effort. One of my three remotes is indeed that one that is branded 'Mi-Light' and I can confirm mine works too. The other two do not however. The colour LED strip one is this one https://www.ebay.com/itm/2-4G-LED-RGB-RGBW-Controller-Touch-Screen-RF-Remote-Control-for-RGBW-LED-Strip-/262983073383?roken=cUgayN&soutkn=huGRwh. Can I send you one of those to try? PM me if that is OK.

sidoh, 2018-05-14:
No need, I think.  I have the milight wifi bridge which is capable of emulating that remote.

I can confirm I'm not seeing any sniffed traffic on my end.

We'll need to get the radio configs (channels, syncwords) to add support for the remote.  [This writeup](https://hackaday.io/project/8758/logs) shows a teardown of a milight device.  My wifi gateway has both of these chips, but I don't have the equipment to do this myself.

Also possible that someone else has done this already.

sidoh, 2018-05-16:
https://www.amazon.com/gp/product/B077LSG5P2

And to think I was halfheartedly considering getting a $400 Saleae.

xbmcnut, 2018-05-17:
Wow, that is awesome! If it did the trick, that is all I need too. Thanks.

sidoh, 2018-05-31:
Just an update here --

This is definitely an issue with the nRF radio code.  I poked around a little bit, but haven't made much progress yet.  (I did, however, manage to get the radio configs for [all remotes supported by the ibox](https://docs.google.com/spreadsheets/d/1eMNn5HmMzNc0LgRBQlcnDYeE7KKCKBPMd15dMbIpqfM/edit#gid=0))

@khmann I recall that some of the radio optimizations you contributed a few months back may have broken sniffing from official devices (this was something we agreed was fine at the time).  Does this sound familiar to you?

If so, any pointers on adding this support back without breaking the radio?  Definitely happy to investigate myself, but thought I'd ask in case you have any ideas offhand.

xbmcnut, 2018-06-05:
Thanks for the update.

xbmcnut, 2018-06-14:
Any NRF changes in the esp8266_milight_hub_nodemcuv2-1.8.0-dev1 build?

sidoh, 2018-07-22:
@xbmcnut hey, sorry I'm so slow getting back to you.  Busy summer!

No, no nRF changes in v1.8.  At the rate I'm going it probably won't make it in.

xbmcnut, 2018-07-22:
Thanks for the update, appreciated. Let us know once you have something I can test.

khmann, 2018-07-23:
Sorry, I dropped off the internet for awhile.  The "bug" relates to the not-clearly-defined PL1167 trailer bits; It is possible that these bits are encoded differently by different transmitters and my original patch (RF mod was to replace the rather silly PL1167 software decoding by using nRF hardware filters) had a problem with some of the codeword calculation.  I believe this was actually fixed in the second patch, but that there was some other problem that broke everything.  I've been meaning to pick this back up (and actually use it this time - previously I was just using the reusing your radio code for my own project)

khmann, 2018-07-24:
Or I'm confused and that's the preamble (I know I verified the preamble function is correct.)  Maybe we need to add the following code (or it's inverse...?)  I'm sorry I'm still trying to figure out platformio to play with it here

[code]
 -  _nrf_pipe[ --nrf_address_pos ] = reverse_bits( ( (_syncword3 >> 12) & 0x0f ) + 0x50 );	// kh: spi doc says trailer is always "5" ?
 +  if (_syncword3 & 0x8000) {
 +    _nrf_pipe[ --nrf_address_pos ] = reverse_bits( ( (_syncword3 >> 12) & 0x0f ) + 0xa0 ); 
 +  } else {
 +    _nrf_pipe[ --nrf_address_pos ] = reverse_bits( ( (_syncword3 >> 12) & 0x0f ) + 0x50 );
 +  }
[/code]

I agree the code needs something, but I think it's worth considering the possibility that some things which _look_ milight compatible, quite simply, aren't.  I've certainly been bitten by that...

xbmcnut, 2018-07-25:
Thanks guys. Let me know if you'd like me to test a couple of different variants with the 'non-working' remotes I have.

xbmcnut, 2018-10-26:
Hi there. Any chance this has been improved in 1.8.5?

hhaensel, 2019-02-24:
Hi there,
I have just tried to sniff commands from my newly acquired FUT098. But no response at all :-(
However, I succeeded in pairing my RGBW controller (FUT027) by chosing an arbitrary device ID, selecting the RGB (!) protocol and pressing `pair` 3s after switching on the controller. I can now control the device via mode/color/brightness/speed.
But also the commands sent from the ESP are not sniffed! So I guess the problem is still  not solved in "1.8.7"?

xbmcnut, 2019-03-03:
Any progress on this issue Chris?

sidoh, 2019-03-03:
Hey, sorry for not responding to your earlier comment, @xbmcnut. Slipped through the cracks.  No, I've honestly not looked at it.  No plans to soon, although if it's important to a number of folks I could prioritize it.

cristianlivella, 2019-04-29:
Same problem here with FUT098 remote and FUT025 controller. I can pair and send comand with NRF24L01, but I can't sniff the remote commands.

sidoh, 2019-04-29:
FUT098 is supported.

cristianlivella, 2019-04-29:
> FUT098 is supported.

It isn't working... during sniffing I see only the packets sent by the NRF24L01.

sidoh, 2019-04-29:
Ah, sorry, missed the context in the thread.

Sniffing doesn't work with it.  Some weird stuff in radio config I haven't had the interest or time to dig into.

Think it's probably somewhere in this change:

https://github.com/sidoh/esp8266_milight_hub/commit/22f3baf667745f6de8887f0482b10073ef0da7e1#diff-955946430a21b8fcd99b02c8913272db

xbmcnut, 2019-05-10:
I'm a patient man. Let us know if you ever get this fixed and I'll give it another try.

xbmcnut, 2019-06-25:
Any of the new builds have a fix for this yet Chris?

sidoh, 2019-06-25:
No, unfortunately not.  If I'm honest, this is not a very fun sort of issue to work on, so sort of hard to drum up the motivation to dig in.

There's probably room in the 1.10 release, but it's possible I won't have the time or motivation to dig through fiddly radio code (apologies for the inconvenience).

xbmcnut, 2019-06-26:
Thanks. I'll move on.

sidoh, 2019-06-26:
I'd like to keep the issue open.  It is definitely an actual bug.

xbmcnut, 2019-06-26:
Understood.

sidoh, 2019-06-30:
Good news... think I figured this out. :)

Should have a patch up later today.

@khmann called it.  I think the PL1167 trailer is different for the RGB remote (the bits are inverted).

xbmcnut, 2019-06-30:
Wow, that's great news. Looking forward to testing.

sidoh, 2019-07-01:
Notes from poking around with the (A) the radio code, and (B) SPI dumps from the wifi gateway:

* The register that defines packet syncword lengths, etc. (`0x20`) is not set, which means the defaults are used:
  * Preamble: 1 byte
  * Syncword: 8 bytes (only 0-2 and 4-6 are set.  others assume the default of 0)
  * Trailer: 4 bits
* There does not seem to be a way to set the values for the preamble or trailer.  I think this means there's something internal to the PL1167 that generates these values.
* Because the values are different based on the syncword, they seem to be a function of the syncword.  (why?  who knows...)
* Seems to depend on the parity of syncword0.
* SPI dump when sending an RGB packet: [rgb_spi.csv.gz](https://github.com/sidoh/esp8266_milight_hub/files/3343910/rgb_spi.csv.gz)
* SPI dump when sending an RGB+CCT packet: [rgb_cct_spi.csv.gz](https://github.com/sidoh/esp8266_milight_hub/files/3343914/rgb_cct_spi.csv.gz)
* SPI dumps ended up not helping much, but attaching them to show that defaults are not changed.
* [PL1167 User Manual](https://github.com/sidoh/esp8266_milight_hub/files/3343915/PL1167_UM_EN_V0.2.pdf)

What ended up revealing what was going on was reducing the nRF24 address with to 4 (from 5), which showed the packet even if the trailer didn't match.  Then you could see what the value of the trailer was for all remote types.  Will hardcode these values into the radio configs.

sidoh, 2019-07-01:
Fixed in [1.10.0-dev.4](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.10.0-dev.4).

xbmcnut, 2019-07-01:
Hi Chris,

Just tried 1.10.0-dev.4 (nodemcuv2) and sadly it is still not sniffing packets from this remote https://github.com/sidoh/esp8266_milight_hub/issues/266#issuecomment-385165682

sidoh, 2019-07-01:
Hm.

I'm testing with the official wifi gateway, and I am able to sniff RGB packets (where I was not able to before).

Is it able to sniff packets from other remotes?

xbmcnut, 2019-07-01:
Yes. My original MiLight remote is being sniffed.

xbmcnut, 2019-07-01:
Sorry, tested [this one](http://futlight.com/productdetails.aspx?id=175&typeid=143) again and sadly, still no go.

sidoh, 2019-07-01:
Double-checked that I'm not going crazy.  The emulated remote is definitely working, and was not working before.  Think yours might be a separate issue.

Any way you can get a SPI bus dump?

sidoh, 2019-07-01:
Any chance this is your remote is one of FUT020, FUT021, FUT022?  These don't work for a different reason (their radio configs aren't present at all).

xbmcnut, 2019-07-01:
My two remotes that don't work aren't any of the above. I'll try connecting another radio and I'll buy a USB debugger for future diagnostics. Anything I can tweak in the web UI?

sidoh, 2019-07-01:
No settings that should affect whether the remote is picked up or not (especially if it's picking up traffic from other remotes).

Nevertheless, could you post the output of `/settings` and `/about` (censoring any passwords)?

Compiling with `DEBUG_PRINTF` and checking whether there's any output while pressing the remote button would be helpful as well.

i found this remote on ebay for $5, but won't be here for a while.

xbmcnut, 2019-07-01:
I'm trying to compile the latest zip in PlatformIO with DEBUG_PRINTF uncommented but getting errors.

`Compiling .pioenvs\d1_mini_pro\lib1d5\RichHttpServer_ID6342\RichResponse.cpp.o
.piolibdeps\WebSockets_ID549\src\WebSocketsClient.cpp: In member function 'void WebSocketsClient::connectedCb()':
.piolibdeps\WebSockets_ID549\src\WebSocketsClient.cpp:749:22: error: 'class axTLS::WiFiClientSecure' has no member named 'setInsecure'
_client.ssl->setInsecure();
^
*** [.pioenvs\d1_mini_pro\libecd\WebSockets_ID549\WebSocketsClient.cpp.o] Error 1
 [ERROR] Took 30.63 seconds
 
 [SUMMARY]
Environment nodemcuv2   [ERROR]
Environment d1_mini     [ERROR]
Environment esp12       [ERROR]
Environment esp07       [ERROR]
Environment huzzah      [ERROR]
Environment d1_mini_pro [ERROR]
 [ERROR] Took 255.75 seconds`

sidoh, 2019-07-01:
Are you using a .zip from one of the v1.10 releases?  It includes a fix for this issue (setting the `USING_AXTLS` flag).

Upstream library fixed the issue (see https://github.com/Links2004/arduinoWebSockets/issues/447), but not sure if it's been released yet.

sidoh, 2019-07-01:
(btw, you can build for a particular platform with, e.g., `pio run -e d1_mini`)

xbmcnut, 2019-07-01:
Apologies, my bad. Downloaded zip from main page. Downloading dev4 now.

xbmcnut, 2019-07-01:
Thanks, seems to be working now with Dev4. Managed to create a .bin for d1_mini. Building nodemcuv2 now.

sidoh, 2019-07-01:
Awesome.  Thanks for pitching in with debugging, much appreciated.

If nothing shows up in serial logs, I'll try uploading an experimental version that doesn't include the trailer in the nRF24 address.

xbmcnut, 2019-07-01:
Happy to test a custom version. Should be able to test Dev4 with serial in next 2hrs.

sidoh, 2019-07-01:
Let me know if anything shows up in serial logs (while you're pressing the remote) first so we can narrow down the problem space.

xbmcnut, 2019-07-02:
Tried serial monitor with Dev4. Working with default MiLight remote [FUT007](http://futlight.com/productdetails.aspx?id=169&typeid=143) but no output at all when using [FUT098](http://futlight.com/productdetails.aspx?id=175&typeid=143) or the remote shown below. Tested another NRF24 module also with the same result.

[image]

xbmcnut, 2019-07-02:
Here is one of the problem remotes on Ali https://s.click.aliexpress.com/e/bdmn23bS

[31 more replies]

PR #483 (2019-07-01): Bugfix: RGB listens with milight remote
Fixes #266 

Radio optimizations added in a much earlier version introduced a bug where RGB commands sent from remotes were not picked up by espMH (RGB commands sent from another espMH instance would be received, however).

This fixes that.  The fundamental issue is that the PL1167 includes a "preamble" and a "trailer" field in the transmitted packet, which appear to be generated as a function of the syncword.  These were hardcoded, but RGB uses a different preamble/trailer.

More detailed notes on the investigation can be found here:
https://github.com/ [... cut]
closed 2019-08-30 by sidoh