sidoh/esp8266_milight_hub #272: Control RGBW Bulb with RGB_CCT Remote (Home Assistant)

sonicnkt, 2018-05-01:
Hey,
just got my milight hub up and running and was in the process of getting everything into Home Assistant. 

I want to use my existing remote to control the lights through home assistant so that the states keep synchronized and maybe use the remote for other non milight devices.

I followed the WIKI to get my test bulb into HA and the demo config for the remote. I have the same remote as the one displayed in the wiki.

Everything workes great except i cant get my RGBW bulb into white mode this way. The Bulb is paired with the hub and not the remote. When i had it paired directly with this remote before i could enable "whitemode" by touching the "white" slider, this does not work anymore. 
In Home Assistant i can set the bulb to white and it also works fine with the milight hub webinterface.
Every other function of the remote works flawlessly out of the box with the example config.

Can someone help me getting this working?

Here is a capture of the mqtt message when i press the remote white slider:
DEVICE_ID 0x34D3 is the Remote
DEVICE_ID 0x1 is the RGBW Bulb paired with my hub

```
milight/updates/0x34D3/rgb_cct/2:
{"color_temp":370}

milight/commands/0x1/rgbw/2
{"color_temp":370}

milight/states/0x34D3/rgb_cct/2
{"state":"ON","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}

```
So the hub is only sending the color_temp command through the update channel, which gets forwarded by home assistant to the bulb. This does nothing as is expected for this bulb. And there is no state change for the 0x1 device. 

Shouldnt this also send {"command":"set_white"} ? 
I probably could but together a python script in HA that looks for the color_temp string and than replace it with {"command":"set_white"} but  i hope there is a better solution. 

If i re-pair the bulb with this remote and ditch the idea of using the remote with other devices (for now) shouldnt it also show the current state correctly in hass. Looking at the mqtt logs shows the correct state for the "remote-device" (0x34D3/rgb_cct/2) which ha should recongize?

Sorry for all the questions, im new to HA, MQTT and all this stuff :)

Im running 1.7.0-dev9 (nodemcuv2) on my hub. 

@sidoh Btw thanks for this great Project! Any way to support your work? (supply new hardware for testing?)

sidoh, 2018-05-01:
RGB+CCT bulbs do switch to white mode when they receive a color temp packet.

If you're setting up MQTT forwarding through HomeAssistant, you could probably pretty easily inject a set_white command if the payload contains color_temp.

sonicnkt, 2018-05-04:
Ok Thanks got it working with some help from the ha community:

```
automation:
  - alias: MiLight Remote - G2
    initial_state: True
    trigger:
      platform: mqtt
      topic: milight/updates/0x34D3/rgb_cct/2
    action:
      - service: mqtt.publish
        data_template:
          topic: "milight/commands/0x1/rgbw/2"
          payload_template: >
            {% if 'color_temp' in trigger.payload -%}
              {{ {"command":"set_white"} }}
            {%- else -%}
              {{ trigger.payload }}
            {%- endif %}
```

With this modification i can completely control my RGBW bulb.