sidoh/esp8266_milight_hub #273: Cannot set wifi on 1.7.0-dev10

bombcheck, 2018-05-02:
Starting with a new, clean flash I am unable to set wifi parameters because of broken web page.
Seems the page requires external JS-libs but can not load them because in this state, the device which I use to configure the hub has no internet connection (it is connected to the hubs access point).

Using a desktop computer with a wifi- and a wired network connection work.

bombcheck, 2018-05-02:
Strange... The WiFiManager should come up in this state... After another power cycle it works (WifiManager shows up when no WiFI is configured). But at first boot after flashing I got the hubs webpage at 192.168.4.1...

closed 2018-05-02 by bombcheck

sidoh, 2018-05-02:
WifiManager times out after 3 minutes.  When this happen looks like it just falls through to normal operation except in AP mode.

I think it should probably reset itself if connection fails.