sidoh/esp8266_milight_hub #274: Control saturation in Home Assistant

kurniawan77, 2018-05-04:
Hi,

How should i configure Home Assistant to control the saturation for RGB+CCT?
This is what i have configured.

[image]
and
[image]

Or must i have something else then white_value?

Regards

sidoh, 2018-05-04:
You can't control saturation from HASS directly, at least not with the `mqtt_json` component.  RGB should expose a color wheel, which will adjust both hue and saturation appropriately behind the scenes:

[image]

If you want to control it directly you'll have to write your own HASS component.

kurniawan77, 2018-05-04:
Thanks for the very very fast reply! The color wheel will do it fine too!

closed 2018-05-04 by kurniawan77