sidoh/esp8266_milight_hub #276: FUT038 Support

salex89, 2018-05-05:
Hi, I hope I'm not misusing the Issue feature here :-) . Does this solution support the FUT038 controller (it looks pretty similar to the FUT025 one). I've got some light strips and a couple of FUT038, so I'm seriously thinking in not buying any hub, but trying to make my own based on this, since I hope it will be better quality. So, just to know the state of it, has anybody tried the FT038?

sidoh, 2018-05-05:
Yes, should work with RGB+CCT.  I've not tried it personally, but the product documentation says it works with the FUT092 remote, which is definitely supported.

> As far as I see it, it is this device (the one you linked in the readme). It is without WiFi support, as in the product description, a hub is needed. The WiFi hub is a separate device, maybe you wanted a link to it instead of this one?

Which link are you referring to?  This project is a replacement for the wifi hub.

> So I flashed the firmware onto a Wemos Lolin V3 board, everything went more or less straightforward, but I'm not sniffing anything for now. I'm using a NRF24L01+ ver A0 module. Now, It's possible I messed something up with the wiring or the module is defective (I have another one). I'll continue to investigate as far as my electronics/embedded knowledge goes :-) .

What device are you expecting to be able to sniff?  If you don't have a remote or a wifi gateway, sniffing isn't going to show anything.

To pair with a new device, follow these instructions:

https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs

salex89, 2018-05-06:
Looks like I overcomplicated my explanation :) .

So, I have RF LED strip controllers (FUT038), without a WiFi Hub or remote control. The FUT038 can work with an FUT092 remote (which I do not have), and this project is a replacement, right? I can use this as a remote?

I obviously misunderstood what is the sniffing used for, and I tried to sniff assuming it is a step preceding pairing. I'll try the pair procedure, thanks.

As for the device confusion. The one I have (FUT038), it is very similar o this: https://www.amazon.com/BTF-LIGHTING-Mi-Light-WiFi-Bridge-Controller/dp/B01H87DYR8/ref=sr_1_7?ie=UTF8&qid=1485715984&sr=8-7&keywords=milight. It is linked in the README as a WiFi gateway, but this is not a WiFi gateway, this is a LED strip controller. In the [blogpost](http://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/), the correct device is linked (the [gateway](https://www.amazon.com/gp/product/B00HCLRBZM/ref=as_li_qf_sp_asin_il_tl?ie=UTF8&tag=chrimullblog-20&camp=1789&creative=9325&linkCode=as2&creativeASIN=B00HCLRBZM&linkId=970b8e32c8183eafd9074322db0a938b)).

salex89, 2018-05-06:
Looks like I overcomplicated my explanation :) .

So, I have RF LED strip controllers (FUT038), without a WiFi Hub or remote control. The FUT038 can work with an FUT092 remote (which I do not have), and this project is a replacement, right?

As for the device I have, it is very similar o this: https://www.amazon.com/BTF-LIGHTING-Mi-Light-WiFi-Bridge-Controller/dp/B01H87DYR8/ref=sr_1_7?ie=UTF8&qid=1485715984&sr=8-7&keywords=milight. It is linked in the README, but this is not a WiFi Hub, this is a LED strip controller. In the blogpost

sidoh, 2018-05-06:
Ah, sorry for the misunderstanding.  I think the product behind the link changed.  I just updated it to point at the futlight.com product list instead.

Yes, you can use it as a remote, but looks like you already figured that out :)

That's all saturation control for any device does -- it mixes white light with colored light.  The more saturated the color, the less white there is.

If you have any suggestions on how to improve color picking in the UI, I'm definitely all ears.  The colors in the color band should correspond pretty well to the colors the bulb display.

salex89, 2018-05-07:
To be honest I'm also not sure. Of course, my device is different than the LED bub so it might have different behavior and gamut.

I'll try to explain what I noticed, which might be expected or not (I'm a newbie in this area).

I've noticed that changing the color using the picker, the web client sends hue, saturation or level. The API responds with the actual RGB value which is actually set (it is not equivalent to the set HSV, but somehow similar/rounded :) ). But when I enter the HSV into a picker like [this](http://hslpicker.com/#7c82da) I and send those RGB values I get a color closer to what I expected and chosen.

Another example, sending H: 0, S: 80, L: 67 gives a much more intensive color than R: 238 G: 103 B: 103, and more intensive (in terms of hue) than it should be. And those two should be near equivalent. Sending the HSL the response is:

```
{
    "state": "ON",
    "brightness": 176,
    "bulb_mode": "color",
    "color": {
        "r": 255,
        "g": 50,
        "b": 50
    }
}
```

and the RGB:
```
{
    "state": "ON",
    "brightness": 176,
    "bulb_mode": "color",
    "color": {
        "r": 255,
        "g": 109,
        "b": 109
    }
}
```
Which is nowhere near, and that shows.

What I wanted to say in the end is using an RGB picker (at least for my setup) and RGB format gives much more accurate colors. Either make a fork with a different color control or just a mobile app since I don't plan using the web that much.

sidoh, 2018-05-31:
@salex89 if I'm understanding correctly we resolved this one through discussion in the PR #279.  If I'm misunderstanding let me know and I can reopen the issue.

closed 2018-05-31 by sidoh