sidoh/esp8266_milight_hub #277: Config-AP (ESPXXXXX) is always visible

bombcheck, 2018-05-06:
Just noticed a strange thing: In my setup, router and miligh hub power cycles once a day. So the hub is back online before the router's wifi is back. The hub later connects fine to the wifi (commands are executes and my bulbs works normal) but the wifi "ESPXXXX" is still visible and can be connected to. Any ideas?

sidoh, 2018-05-06:
I think this should fix:

https://github.com/sidoh/esp8266_milight_hub/pull/278

Are you able to test?

bombcheck, 2018-05-07:
The hub is on an external installation. Just did a test: Issues still persists. Have another ESP with ESPeasy: Here the config ap disappears after some time when connection to router wifi is established. But the MilightHub's config ap is always visible.

sidoh, 2018-05-07:
Note that this does mean that you have to give it at least 3 minutes after booting (to be safe, after it appears on the network) for the config AP to go away.

bombcheck, 2018-05-07:
The config ap stays there forever. When I manually reboot the hub (while the router wifi is there) the ap disappears.

sidoh, 2018-05-07:
When you say "manually reboot," do you mean physically press the reset button?  Does it work if you reset via the API or UI?

bombcheck, 2018-05-07:
Physically turn off/on the power. A reboot via gui/api is not enough. But I am not 100% sure about that.

sidoh, 2018-05-08:
Would be good to get some serial logs.  I can get some myself as well, but won't be able to until later in the week

sidoh, 2018-05-09:
Latest code in the PR should fix it

bombcheck, 2018-05-09:
Just checked: Bug is fixed. Thx!

sidoh, 2018-05-12:
Released in 1.7.1.