sidoh/esp8266_milight_hub #282: Can't save Device id

itProfi, 2018-05-18:
Version 1.7.0 with fut_091_support. When i put in field ane correct Device id - it Entered, but not saved...
When i put manually Device id in Dec format in settings.json - and restore it - all works fine..
[image]
[image]

bombcheck, 2018-05-18:
Yeah noticed that, too. With older version you had to submit setting (bottom of page) the save the device id. With new version just click on "Settings" and then on "Submit".

sidoh, 2018-05-18:
Yep, @bombcheck is right.  Have to save the settings.

No reason for it to not auto-save, though.  We can leave this ticket open for that feature

QuagmireMan, 2018-06-01:
I am having the same issue, but even going into the settings and pressing submit does not solve the issue.

closed 2019-05-28 by sidoh