sidoh/esp8266_milight_hub #283: Feature Request : Set a default off state

zestyphresh, 2018-05-23:
Would it be possible to have the option to set any (all) bulbs to a fixed low brightness (say 5%) before the 'off' packets are sent? I'm just trying to think of a solution to the fraction of a second it takes a bulb to change to the desired brightness from the previous state when turned off, which often means a bright flash followed by the desired state when using evening scenes/lighting. 

I'm using Home-Assistant and i've got a few ideas how to implement this there (any 'off' scene actually does the setting to low brightness and then a node-red automation calls the off command from the scene event) but it might be a useful feature for others as well.

sidoh, 2018-05-23:
I think this has come up before, but I forget where.

My inclination is that this shouldn't be a native feature, and that the API allows for upstream pieces to implement this pretty easily (as you're getting at in your second paragraph).

Willing to believe I'm wrong, but I don't want to overcomplicate espMH, and this feels like a step in that direction.

sidoh, 2018-05-31:
@zestyphresh -- I'm going to close this, but don't take that as a "no."  Definitely happy to discuss further, just want to keep the backlog clean. :)

closed 2018-05-31 by sidoh

jespertheend, 2019-12-14:
Hi, I'd like to see this feature built into the api. And I have a reason why I think this might be better as a feature in the current api, rather than upstream pieces having to implement this.

Basically I'm using [homebridge-milight-hub](https://github.com/shamasis/homebridge-milight-hub) for homekit support. I tried implementing this feature in there first, but it is using the state of the lights returned from the esp api. This means that when you send a `{brightness:1, state:"off"}` command to the esp, the brightness gets updated as well.
So when you want to turn the light on again, it will always have the lowest brightness, rather than the brightness it had when you turned it off.
One way to circumvent this would be to keep track of a separate brightness state in any upstream pieces, but this would mean that if another system also uses the web api of the esp, it wouldn't sync up properly with whatever system is turning off lights *and* also setting the brightness to 1.

I'm still investigating a bit on what would be the best way to implement this, but my current proposal would be to add an extra setting below the `Switch to previous mode after saturation/color commands` in the `radio` tab. Something like `Send brightness commands when turning lights on or off.`
Then, either in `MiLightHttpServer::handleRequest` make sure the call to `milightClient->update(request);` contains brightness data. Or in `MiLightClient::update` itself make sure the appropriate `updateBrightness()` calls get made. Either set the brightness to 1 one when turning off. Or set it to the last brightness state when turning on. Only when the setting is enabled obviously. I'm not really sure yet how to prevent the new brightness updates from also updating the state though. I haven't figured out exactly where this is happening.

Let me know if you think this is a good idea or if you'd rather have me implement this in [homebridge-milight-hub](https://github.com/shamasis/homebridge-milight-hub) after all. Also if you know of a better way to implement it let me know. I'd be happy to do a PR but help would be appreciated since I'm not too familiar with this code.