sidoh/esp8266_milight_hub #284: Group 0 packets should not clobber state for individual groups

sidoh, 2018-05-25:
## Summary

When a packet for group 0 is received, we update the state for group 0, and then overwrite the states for individual groups.

Instead, the incremental update should be applied to each of the individual groups, and group 0 should not have a state (it's meaningless since it's a "virtual" group).

## Notes

* [ ] Apply group 0 increments to individual groups, rather than overwriting with group 0 state.
* [ ] We should still have an option to send MQTT states for group 0
* [ ] We should consider adding an option to send fanned out MQTT states.  While this is heavy-weight, the alternative is to force users to merge states themselves, which is also not ideal.

## Reproduce the issue

#### Initialize state to "on" and brightness to max for virtual group 0
```
$ curl -H 'content-type: application/json' -X PUT -d '{"status":"on","level":100}' 10.133.8.158/gateways/1235/rgbw/0
{"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}%

$ curl 10.133.8.158/gateways/1236/rgbw/\{1,2,3,4\}

[1/4]: 10.133.8.158/gateways/1236/rgbw/1 --> <stdout>
--_curl_--10.133.8.158/gateways/1236/rgbw/1
{"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}
[2/4]: 10.133.8.158/gateways/1236/rgbw/2 --> <stdout>
--_curl_--10.133.8.158/gateways/1236/rgbw/2
{"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}
[3/4]: 10.133.8.158/gateways/1236/rgbw/3 --> <stdout>
--_curl_--10.133.8.158/gateways/1236/rgbw/3
{"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}
[4/4]: 10.133.8.158/gateways/1236/rgbw/4 --> <stdout>
--_curl_--10.133.8.158/gateways/1236/rgbw/4
{"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}%
```

#### Set brightness to 10% for group 1
```
$ curl -H 'content-type: application/json' -X PUT -d '{"status":"on","level":10}' 10.133.8.158/gateways/1236/rgbw/1
{"state":"ON","brightness":31,"color":{"r":255,"g":255,"b":255}}%
```

#### Turn group 0 on (notice we're not setting brightness)
```
$ curl -H 'content-type: application/json' -X PUT -d '{"status":"on"}' 10.133.8.158/gateways/1236/rgbw/0
{"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}%
```

#### State for group 1 is clobbered by group 0 state
```
$ curl 10.133.8.158/gateways/1236/rgbw/1
{"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}%
```

boaschti, 2018-06-07:
Hi, great work. Can you built .bin?

sidoh, 2018-06-08:
Sure.  Let me get the FUT091 support merged as well first.  Should have time this weekend.  If not I'll release a version with just this fix.

sidoh, 2018-06-12:
Tagged 1.8.0-dev1 with this fix.  Travis should publish the binaries shortly

commit 80c84de6 (2018-05-31): Apply patched updates to individual groups when updating group 0 (fixes #284) (#287)

* Set up unit tests

* Add method to patch state with set fields from another state

* add methods to help with test

* Patch member state with virtual "all groups" state instead of clobbering it

* Don't persist group 0 state, except for devices that don't have groups

* Stateful brightness for RGB
closed 2018-07-29 by sidoh