sidoh/esp8266_milight_hub #286: sonoff

agoralive, 2018-05-28:
Hi

Is there a waz to use the sonoff TH10 with this build?

I have a few of these which have been "modified" with the Tasmota script, and I would like to use these perhaps for my Mi-Light system....?

sidoh, 2018-05-28:
The only real difference in the builds is that they're compiled for different flash sizes.  nodemcuv2 and d1_mini are pretty much interchangeable, for example.

As long as your chip has SPI pins exposed, and you can attach an nrf24l01, you should be able to use one of the existing builds.

If it has 4 MB flash, d1_mini should work.  If 1 MB, esp07 should work.

sidoh, 2018-05-29:
Going to close, but feel free to reopen if you feel like your question wasn't addressed.

closed 2018-05-29 by sidoh