sidoh/esp8266_milight_hub #293: As it is possible to compile the web index.html.gz.h file simple means or the help to the beginner

vsv78vrn, 2018-06-12:
Hello dear colleagues. I am a beginner in such questions. I have a big question? How it is possible to add in a web the interface the features? Or what easy ways to reform the index.html.gz.h file???)))
Is where be an opportunity to esteem documentation? Or how he can be reformed? I work in Windows 10. Already I struggle the 2nd week with this question. If it is possible on what be the simple instruction for beginners!
Thanks in advance! Your project the best)))

sidoh, 2018-06-12:
The platformio build should work just fine on Windows so long as you've got node and gulp installed.

Here are the Windows-specific lines that get run:

https://github.com/sidoh/esp8266_milight_hub/blob/master/.build_web.py#L23-L25

vsv78vrn, 2018-06-12:
and here that's just the point what doesn't work and why I can't understand!))) what distribution kits need to be established still?

vsv78vrn, 2018-06-12:
Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:25:58) [MSC v.1500 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
======== RESTART: C:\Arduino\esp8266_milight_hub-master\.build_web.py ========

Traceback (most recent call last):
  File "C:\Arduino\esp8266_milight_hub-master\.build_web.py", line 8, in <module>
    Import("env")
NameError: name 'Import' is not defined
>>>

vsv78vrn, 2018-06-12:
How to recompile the file index.html.gz.h hands?

vsv78vrn, 2018-06-12:
Terminal in Atom

sidoh, 2018-06-12:
You can't run that script directly.  It's run as a part of the platformio build (eg `pio run -e d1_mini`).

Running `gulp` in the `./web` directory will work if you have node and gulp installed.

sidoh, 2018-06-13:
I have not made an attempt to document setting up a local build environment, but it's probably worthwhile to do so.

sidoh, 2019-04-07:
Added a [Development section](https://github.com/sidoh/esp8266_milight_hub/tree/1.9.0#development) to the README.

closed 2019-04-07 by sidoh