sidoh/esp8266_milight_hub #296: Range on FUT103

alekslyse, 2018-06-18:
This might be a bit bulb specific compared to software fault, but not sure. I setup the software on a nodeMCU with a NRF24L01+PA+LNA SMA (The one with antenna).

It works great with the  FUT014 bulbs (regular round bulbs), but with the FUT103 (GU10) I cant get any range out of it. The FUT014 works with no issues outside the house (about 20m from the ESP), and the FUT103 is 1m away from them, refusing to do anyting. Moving them 1-2m from the ESP they work.

Anyone else with some thoughts about range and FUT103?

sonicnkt, 2018-06-22:
I have a few FUT103 and i noticed something similiar but not as bad as yours.
They work fine with the nodemcu board 7-10m away and in another room but i have to add that i have no brick walls in my apartment. 
Putting the hub in the center of my flat works fine to controll everything here (85m²).

i could imagine that the range also gets influenced by the gu10 lamp/socket. If its a full metal housing that covers the whole bulb with only the front exposed it could be worse?

sidoh, 2018-06-23:
Do your FUT103s work better with official milight devices?

You could try cranking up the number of repeats, but probably shouldn't be necessary.

sidoh, 2018-07-22:
Gonna close because it seems unlikely that it's a software issue.  But please feel free to re-open if you have data that suggests otherwise.

closed 2018-07-22 by sidoh