sidoh/esp8266_milight_hub #297: ESP12E or Wemos PRO min reboots when pressing the button of adjustment of a cold-warm luminescence from the FUT089 panel

vsv78vrn, 2018-06-22:
Radio is available

Packet received: 909C9ECF41DEBF724163Packet transformed: 093979F3827BFD4E82C6Failed CRC: expected 33540, got 33478
Radio is available

Packet received: 909C9ECF41DEBF724163Packet transformed: 093979F3827BFD4E82C6Failed CRC: expected 33540, got 33478
Radio is available

Packet received: 909C9ECF41DEBF724163ADB3Packet transformed: 093979F3827BFD4E82C6B5CDSuccessfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 14790

Exception (28):
epc1=0x40218dd3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: cont 
sp: 3fff1eb0 end: 3fff2320 offset: 01a0

>>>stack>>>
3fff2050:  3ffeb100 401042a0 3ffee9b0 40100eb6  
3fff2060:  3fff35cc 00000000 3fff2280 40218e10  
3fff2070:  00000005 00000000 00000020 40218bd9  
3fff2080:  3ffeb105 401042bb 3ffeeb68 00000000  
3fff2090:  40101bd1 3fff2280 3fff20e0 40217397  
3fff20a0:  5c552539 01096407 3ffef5cc 40220764  
3fff20b0:  3ffe8308 3ffe8308 3fff47f8 3fff09e4  
3fff20c0:  3fff09e0 3fff0ed8 3fff2280 402081cf  
3fff20d0:  40106e40 3fff20e0 000000c8 00000008  
3fff20e0:  3fff20d0 00000000 3fff2110 3fff47f8  
3fff20f0:  40106a50 3ffe8308 3fff47f8 4022b140  
3fff2100:  3ffe8308 3ffe8308 3fff47f8 4022b140  
3fff2110:  3fff4934 3ffe8308 3fff47f8 00000011  
3fff2120:  3fff4934 3ffe8308 3fff47f8 4022b5b1  
3fff2130:  3ffe8308 3ffe8308 3fff2160 3fff47f8  
3fff2140:  0000000a 3ffe8308 3fff47f8 4022b641  
3fff2150:  4022acb4 3ffe8308 3fff47f8 4022b26b  
3fff2160:  3ffe9ae2 3ffe9ae1 00000000 4022acab  
3fff2170:  3ffe8308 00000001 0000000a 4022ad04  
3fff2180:  3ffe8308 3fff2b60 3fff3790 3fff47f8  
3fff2190:  3ffe9ba5 00000000 3ffe8308 3fff47f8  
3fff21a0:  3ffe9ae1 00000000 3ffe8308 4022ae84  
3fff21b0:  00000000 ffffffff ffffffff 00000000  
3fff21c0:  00000005 00000010 00302064 00000002  
3fff21d0:  3ffe9b74 00000001 3fff2b60 00001809  
3fff21e0:  00303107 3fff435e 3fff4350 34319eb8  
3fff21f0:  00303937 3ffe9b4c 00000000 40226ee2  
3fff2200:  3fff2260 3fff2250 00000000 00000002  
3fff2210:  3fff2270 3fff2260 00000008 00000005  
3fff2220:  3ffe9ae2 00000001 3fff4350 3fff09e4  
3fff2230:  3fff4350 3ffe9b8c 3ffe8308 3fff09e4  
3fff2240:  3fff4350 3ffe9ad4 3ffe8308 40226ee2  
3fff2250:  3fff2270 3fff2260 00000004 00ff0000  
3fff2260:  3ffe9b74 000039c6 3fff35cc 402169d0  
3fff2270:  5c552539 01096407 3fff43cc 402132a5  
3fff2280:  3f01555c 00000004 00000010 402157d9  
3fff2290:  3fff22c0 3fff22c0 3fff20e0 402249c8  
3fff22a0:  00000009 3fff434c 00000002 3fff09e4  
3fff22b0:  3fff09e0 3fff434c 00000002 4020834d  
3fff22c0:  82f37939 824efd7b 000013c6 3fff12f8  
3fff22d0:  00000009 00000000 3fff50ec 4021dbe5  
3fff22e0:  3fffdad0 00000000 3fff46ac 3fff12f8  
3fff22f0:  3fff09bc 3fff09e4 00000000 402083c7  
3fff2300:  3fffdad0 00000000 3fff12f1 4022050c  
3fff2310:  feefeffe feefeffe 3fff1300 40100700  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
MqttClient - Connecting to: m20.cloudmqtt.com:18217
MqttClient - connecting
MqttClient - subscribing to topic: milight/+/+/+
MqttClient - Successfully connected to MQTT server
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.182
Wifi connected succesfully

Setup complete (version UNKNOWN)

sidoh, 2018-06-22:
What version are you using?

vsv78vrn, 2018-06-22:
Good afternoon! I tried esp8266_milight_hub-1.7.0, esp8266_milight_hub-1.7.1, esp8266_milight_hub-1.8.0. Only this button freezes up.

vsv78vrn, 2018-06-22:
from a web of the interface and team on MQTT are fulfilled! Everything is normal! hangs at reception of team from the panel. When pressing the slider

vsv78vrn, 2018-06-24:
#include <GDBStub.h>
=

Linking .pioenvs\nodemcuv2\firmware.elf
c:/users/viking/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pioenvs\nodemcuv2\firmware.elf section `.text' will not fit in 
region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\nodemcuv2\firmware.elf] Error 1
 [ERROR] Took 6.15 seconds

vsv78vrn, 2018-06-24:
How to catch a mistake?

sidoh, 2018-06-24:
What are you trying to do here?  Are you adding libraries?

Sounds like the compiled firmware won't fit.  There's not really anything you can do except trim down the code.

vsv78vrn, 2018-06-24:
No! I try in any ways to find a mistake which leads to lag.

vsv78vrn, 2018-06-24:
I tried to connect debugging. Nothing good left.

vsv78vrn, 2018-06-24:
yo

vsv78vrn, 2018-06-24:
video in google
https://drive.google.com/open?id=1UxrXA33Sq1em3oIU95OfjONUb0alZ4l4

sidoh, 2018-06-24:
Thanks for the video.

I'm having a hard time reproducing this.  I tried:

* Controlling from the UI / REST / MQTT
* Sniffing packets from iBox 1 milight remote for FUT089
* Sniffing packets from a FUT092 remote

I don't have a physical FUT089 remote, unfortunately.

Could you:

* Get a fresh stacktrace when it crashes
* Upload your `firmware.elf`.  e.g.:
   ```
   $ ll .pioenvs/d1_mini/firmware.elf
   -rwxr-xr-x 1 cmullins staff 1.3M Jun 23 11:30 .pioenvs/d1_mini/firmware.elf
   ```

We can decode the stacktrace with this.

Have you already tried a pre-compiled binary? 

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.7.1

vsv78vrn, 2018-06-24:
Have you already tried a pre-compiled binary?
https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.7.1
Yes ))) all the same hangs

I haven't understood a little what needs to be made with firmware.elf
I will make everything that you will tell)))
forgive but I am a beginner... I understand not everything from the first)))

vsv78vrn, 2018-06-25:
https://drive.google.com/open?id=1eWDudleZCjEBR9cWzGdRsRlQgOtg_Zor

vsv78vrn, 2018-06-25:
Initial codes of working decoding. I don't know a source.
https://drive.google.com/open?id=1VfPby-8oAbOy7RPKsH3BJzIB3Edbx4Mk

new video 
https://drive.google.com/open?id=1eWDudleZCjEBR9cWzGdRsRlQgOtg_Zor

It can do as be to help?

vsv78vrn, 2018-06-26:
Kind to you time of day! And if to buy you the panel given fut089! Will you be able to correct these lags? I am ready to send money for PayPal for his purchase. )))

vsv78vrn, 2018-06-26:
I found! the file - FUT089PacketFormatter.cpp 
THE MAIN LINE OF THE ERROR IN LINE!
    GroupState& state = stateStore->get(bulbId);
I commented out lines!
    //GroupState& state = stateStore->get(bulbId);
    //if (state.getBulbMode() == BULB_MODE_COLOR) {
      //Serial.println("52");
      //result["saturation"] = 100 - constrain(arg, 0, 100);
    //} else {

      result["color_temp"] = Units::whiteValToMireds(100 - arg, 100);

    //}
in the procedure

BulbId FUT089PacketFormatter::parsePacket(const uint8_t *packet, JsonObject& result) {
  uint8_t packetCopy[V2_PACKET_LEN];
  memcpy(packetCopy, packet, V2_PACKET_LEN);

  V2RFEncoding::decodeV2Packet(packetCopy);

  BulbId bulbId(
    (packetCopy[2] << 8) | packetCopy[3],
    packetCopy[7],
    REMOTE_TYPE_FUT089
  );

  uint8_t command = (packetCopy[V2_COMMAND_INDEX] & 0x7F);
  uint8_t arg = packetCopy[V2_ARGUMENT_INDEX];
  
  if (command == FUT089_ON) {
    Serial.println("20");
    if ((packetCopy[V2_COMMAND_INDEX] & 0x80) == 0x80) {
      result["command"] = "night_mode";
    } else if (arg == FUT089_MODE_SPEED_DOWN) {
      result["command"] = "mode_speed_down";
    } else if (arg == FUT089_MODE_SPEED_UP) {
      result["command"] = "mode_speed_up";
    } else if (arg == FUT089_WHITE_MODE) {
      result["command"] = "set_white";
    } else if (arg <= 8) { // Group is not reliably encoded in group byte. Extract from arg byte
      result["state"] = "ON";
      bulbId.groupId = arg;
    } else if (arg >= 9 && arg <= 17) {
      result["state"] = "OFF";
      bulbId.groupId = arg-9;
    }
  } else if (command == FUT089_COLOR) {
    uint8_t rescaledColor = (arg - FUT089_COLOR_OFFSET) % 0x100;
    uint16_t hue = Units::rescale<uint16_t, uint16_t>(rescaledColor, 360, 255.0);
    result["hue"] = hue;
  } else if (command == FUT089_BRIGHTNESS) {
    uint8_t level = constrain(arg, 0, 100);
    result["brightness"] = Units::rescale<uint8_t, uint8_t>(level, 255, 100);
  // saturation == kelvin. arg ranges are the same, so can't distinguish
  // without using state
  } else if (command == FUT089_SATURATION) {

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
    //GroupState& state = stateStore->get(bulbId);
    //if (state.getBulbMode() == BULB_MODE_COLOR) {
      //Serial.println("52");
      //result["saturation"] = 100 - constrain(arg, 0, 100);
    //} else {
      result["color_temp"] = Units::whiteValToMireds(100 - arg, 100);
    //}
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

  } else if (command == FUT089_MODE) {
    Serial.println("60");
    result["mode"] = arg;
    Serial.println("61");
  } else {
    result["button_id"] = command;
    result["argument"] = arg;
  }
  return bulbId;
}

vsv78vrn, 2018-06-26:
I will write result on video soon )))))

sidoh, 2018-06-26:
I just ordered a FUT089, but will take a few weeks to get here from China.

`firmware.elf` is generated when you compile from source.  Could you try this: [firmware.bin.zip](https://github.com/sidoh/esp8266_milight_hub/files/2138861/firmware.bin.zip)?  I have the `.elf` for this.

Just a heads up -- it'll probably be a few days between responses.

sidoh, 2018-06-26:
And just to be clear, running with the firmware above, please provide a stacktrace like you did in your original post on this issue (`Exception (28)` down to `ets Jan 8 2013,rst cause:2, boot mode:(3,6)`)

vsv78vrn, 2018-06-27:
I loaded your firmware.bin. All super! There are no hangups. There are no resets.
it was written Setup complete (version 1.7.1)

vsv78vrn, 2018-06-27:
After long search I it solved problems. As wrote above where a problem.
I recorded video as I comment a code.
video
https://drive.google.com/open?id=1ZdBXh1jrOIAvLCAqXoL70bMw605bsrBK

sidoh, 2018-06-30:
Hey @vsv78vrn, just had time to watch your video.  This is super helpful.  Thanks for finding the line causing the crash.

vsv78vrn, 2018-06-30:
Good luck and progress to you in your project! Excellent project! I will help further you with development of this project)))

sidoh, 2018-07-14:
Got the remote, and was able to reproduce the issue.  Pretty strange that it doesn't come up with iBox.

Don't have time to dig into what's going on right now, but hopefully will over the next couple of days.

sidoh, 2018-07-23:
The problem was `PacketFormatter#prepare` is never called by the branch of code that handles listens.  This means `stateStore` is never set and we're basically dereferencing a null pointer.

There are probably two fixes here:

1. Construct instances of PacketFormatter with a reference to stateStore
2. Make methods that reference `stateStore` take it in as an argument
3. Make sure `prepare` is called in the listen branch

I feel that (2) is the preferable option here, but the patch is (3) because it's the easiest.

@cmidgley do you recall why we removed `stateStore` from `PacketFormatter#parsePacket` in this PR?

https://github.com/sidoh/esp8266_milight_hub/pull/227/files#diff-38661d39e1320654e10bcf3aa2871ab1L49

cmidgley, 2018-07-23:
I believe the answer is that we moved stateStore from a property of the method to become a property of the object.  So "stateStore" is still available, just no longer passed in.  Is this what you are asking?

sidoh, 2018-07-23:
Yeah, just wondering why we decided to make the change.

The issue with passing it in is that it's possible for it to be unset.  There's no compile-time guarantee that the appropriate method to set it has been called.

I think that's more error prone than explicitly passing it in either in the constructor, or in the methods that need access to it.  The bug described in this ticket happened because `prepare` was not called in the code that handles listening.

I think it'd be difficult to pass them into the constructor right now.  Would basically have to layer things in factories.  So passing it into the methods is probably easier.

Just wondering if there was a reason we removed it other than saving callers the effort of passing it in.

cmidgley, 2018-07-23:
It was a case of having to hack it into a large number of methods to get it passed "low enough" where it was needed.  Rather than changing a ton of method signatures just to pass it down, I thought it was cleaner to make it a property.  

An alternative might be to switch it to be a reference to another object that we can query to get state, and use it as a state container.  Just an idea.

sidoh, 2018-07-23:
Thanks Chris!  That helps me understand.  :)

sonicnkt, 2018-07-28:
Just noticed this as well with my FUT089 and NODEMCUv2 (official latest release).
I dont have the fut089 paired with any device but have a big python script in HA to forward commands to different remotes, turn on switch or send ir packets using a rmmini. 

Now i noticed that color_temp isnt working and the remote stops working for a few seconds if i try to change it. (Thats when it reboots i guess) 
I swear it was working before but i could have been paired with a real bulb at this time. 

Anyway, thanks for fixing the problem !

commit a4eb74ad (2018-07-22): Fix #297
commit f117c018 (2018-07-22): Merge pull request #308 from sidoh/fix_fut089_bug

Fix #297
closed 2018-07-29 by sidoh