sidoh/esp8266_milight_hub #300: RGB lights with Flux component - color is reversed

oakbrad, 2018-06-25:
Hi, I am not sure if this is an issue with the Home Assistant Flux component, or something else. I noticed Flux supports RGB bulbs now, so I enabled it for my RGB bulbs. The color changes, but it appears to be running in reverse - ie. warm during the day.

Here is my Home Assistant config
```
- platform: flux
  lights:
    - light.stove
    - light.fridge
    - light.dining_floor_lamp
    - light.floor_lamp
    - light.globes
    - light.reading_lamp
    - light.middle_pendant
    - light.office_pendant
    - light.entry
  name: Flux RGB
  start_time: '5:00'
  stop_time: '10:35'
  disable_brightness_adjust: True
  mode: rgb
```

And then in MQTT I see the Flux component issuing color values like this:
```
lights/updates/0x8000/rgbw/2 {"state":"ON"}
lights/updates/0x8000/rgbw/2 {"hue":26}
lights/states/0x8000/rgbw/2 {"state":"ON","status":"ON","brightness":255,"level":100,"bulb_mode":"color","color":{"r":255,"g":106,"b":0}}
lights/updates/0x2000/rgbw/4 {"state":"ON"}
lights/updates/0x2000/rgbw/4 {"hue":26}
lights/updates/0x8000/rgbw/1 {"state":"ON"}
lights/updates/0x8000/rgbw/1 {"hue":26}
lights/states/0x2000/rgbw/4 {"state":"ON","status":"ON","brightness":41,"level":16,"bulb_mode":"color","color":{"r":255,"g":106,"b":0}}
lights/updates/0x8000/rgbw/3 {"state":"ON"}
lights/updates/0x8000/rgbw/3 {"hue":26}
lights/states/0x8000/rgbw/1 {"state":"ON","status":"ON","brightness":255,"level":100,"bulb_mode":"color","color":{"r":255,"g":106,"b":0}}
lights/states/0x8000/rgbw/3 {"state":"ON","status":"ON","brightness":235,"level":92,"bulb_mode":"color","color":{"r":255,"g":106,"b":0}}
```

Which sets a warm color obviously, although it's noon here currently so I would expect to see a cool color. Any advice for troubleshooting?

sidoh, 2018-06-26:
I would really hope there's nothing in espMH misbehaving since it's just responding to RGB colors.

Is it possible the timezone is set to something unexpected on the machine HASS is running on?

sidoh, 2018-07-14:
@oakbrad were you able to figure out what caused this?

sidoh, 2018-07-22:
I'm gonna close this since it seems unlikely that it's an issue with espMH.  Feel free to reopen if you think otherwise.

closed 2018-07-22 by sidoh