sidoh/esp8266_milight_hub #301: Add support to LS4 dimmer

silviob72, 2018-06-30:
Hi,
Is possible to use this gateway with an LS4 dimmer form Mi Light?

sidoh, 2018-06-30:
From what I can tell, LS4 supports both the FUT006 and FUT007 remote.  If that's the case, you can use the "CCT" remote type.  If that doesn't work, feel free to re-open.

closed 2018-06-30 by sidoh

silviob72, 2018-06-30:
Thanks for the reply, maybe I have another problem.
I use a wemos d1_mini with esp32 and a NRF24L01 module. I have checked many times the wiring and I have tried to replace the nrf24 radio and the d1 too. But when I try to pair, the log in serial is (every time):

```
Soft WDT reset

ctx: cont
sp: 3fff1850 end: 3fff1e40 offset: 01b0

>>>stack>>>
3fff1a00:  00000000 3fff2940 3fff2680 40221218
3fff1a10:  0000000a 000000ff 3fff2940 40221258
3fff1a20:  0000000a 000000ff 3fff2940 402214b7
3fff1a30:  0000000a 00000030 3fff2940 40221510
3fff1a40:  0000000a 00000030 3fff2940 4022162e
3fff1a50:  3fff3e50 00000008 3fff3e50 40221673
3fff1a60:  3fff6ab4 00000037 3fff3e50 402130b6
3fff1a70:  84005ae0 d1001080 00003187 3fff648c
3fff1a80:  00000b68 0000057d 00000000 3fff205c
3fff1a90:  00000001 00008ce1 0000000a 00000009
3fff1aa0:  ffffffc0 0000000a 3fff3e59 401066bd
3fff1ab0:  401065c4 3fff2680 3fff2940 3fff3e9a
3fff1ac0:  3fff3e50 3fff3e4c 00000001 40212cf9
3fff1ad0:  3fff5e18 00000004 3fff1b60 3fff3e4c
3fff1ae0:  3fff3024 3fff3e4c 00000007 40212cb0
3fff1af0:  3fff5e08 00000001 3fff3d84 40213bf6
3fff1b00:  40212c8c 3fff311c 3fff311c 402161bc
3fff1b10:  00000000 3fff3e5e 3fff3d84 3fff3d84
3fff1b20:  3fff5e08 3fff3134 3fff3d84 40213c98
3fff1b30:  3fff5e08 3fff1b60 3fff3d84 40213f58
3fff1b40:  00000000 3fff44ac 3fff3d84 40214089
3fff1b50:  3fff5e08 3fff44ac 3ffe9ae8 402145ac
3fff1b60:  3fff4134 0000000f 00000004 402240bc
3fff1b70:  3ffe8710 3fff1bd0 00000001 40213ae4
3fff1b80:  000000ff 00000312 00000312 00000001
3fff1b90:  00000021 3fff08cc 3fff3d84 3fff1bf0
3fff1ba0:  3fff44c4 3fff44ac 3fff08cc 40219104
3fff1bb0:  3fff44c4 3fff44ac 3fff08cc 4021a435
3fff1bc0:  00746363 3fff44ac 3fff1c70 4021f1fa
3fff1bd0:  3fff0031 007de9ae 00000007 3fff1bf0
3fff1be0:  3f003333 3fff44ac 00000000 4021a2de
3fff1bf0:  00000000 00000000 00000000 00000022
3fff1c00:  3ffeb690 007de9ae 00002200 4000050c
3fff1c10:  3fffc278 40101f80 3fff1bc0 00000000
3fff1c20:  00000003 0000002c 00000003 3fff1bd0
3fff1c30:  00000000 00000001 0000062c 00000001
3fff1c40:  3fff1be0 00000000 00000002 3fff5b2c
3fff1c50:  00000002 40107070 3fff447c 3fff5dfc
3fff1c60:  00000200 00000000 3fff447c 4021f17b
3fff1c70:  3fff411c 0000000f 00000003 3fff4104
3fff1c80:  0000000f 00000001 3fff441c 0000000f
3fff1c90:  00000002 3f000000 000000ff 3fff205c
3fff1ca0:  00000000 00000021 3fff1bf0 3fff5e08
3fff1cb0:  00000001 3fff1bc0 3fff1d10 4021f12f
3fff1cc0:  74616700 79617765 33330073 3fff1d30
3fff1cd0:  3fff1d80 3fff5aec 3fff5404 4021a4f4
3fff1ce0:  3fff1cc0 3fff5aec 3fff5404 4021a82c
3fff1cf0:  3fff1d44 0000030c 0000030c 4010020c
3fff1d00:  3fff1d80 00000001 3fff4a94 4021aca8
3fff1d10:  74616700 79617765 33330073 74636300
3fff1d20:  3f003100 4021ab54 3fff4a94 4021ac4e
3fff1d30:  3fff1d10 3fff1d21 00000012 4021f12f
3fff1d40:  00000010 3fff541c 3fff1d30 4021f1ad
3fff1d50:  3fff1d10 3fff4508 3fff1d80 4021f2b0
3fff1d60:  00000000 00000000 3fff44c4 4021ac18
3fff1d70:  3fff4a94 3fff4508 3fff44c4 4020b795
3fff1d80:  3fff5aec 0000001f 00000012 00000000
3fff1d90:  00000000 00000000 3fff44c8 40209258
3fff1da0:  00000012 00000001 00000000 00000001
3fff1db0:  00000003 3fff5b04 00000012 3fff4a94
3fff1dc0:  00000000 3fff5cac 3fff44c4 3fff0e0c
3fff1dd0:  00000001 3fff44ec 3fff44c4 4020b9d4
3fff1de0:  40106f10 00000000 00001388 3fff0e0c
3fff1df0:  00000000 3fff5cac 3fff45c0 4020f84c
3fff1e00:  3fffdad0 00000000 3fff44ac 40218f18
3fff1e10:  00000000 00000000 3fff0e05 4020836e
3fff1e20:  3fffdad0 00000000 3fff0e05 4021fcd8
3fff1e30:  feefeffe feefeffe 3fff0e20 40100700
<<<stack<<<
           ▒▒▒▒D▒J▒*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.100.111
Wifi connected succesfully

Setup complete (version UNKNOWN)

```

Do you have any idea how to solve the problem?
Thanks