sidoh/esp8266_milight_hub #304: Maximum frame rate accepted on UDP?

mariusmotea, 2018-07-11:
Hi,

I want to extend Hue Entertainment api to MiLight devices, but i don't know if MiLight Hub is able to process UDP frames at 25Hz and forward them to the lights. Did anybody tried the maximum capabilities of the MiLight Hub on UDP channel?

sidoh, 2018-07-14:
Hi @mariusmotea,

Don't have exact numbers, but 25 Hz is probably a best pushing the boundaries, and is more likely beyond what can be expected.

There are some things we could do to improve that (mainly making everything asynchronous), but that's a pretty large undertaking.

There are some performance tuning tips here, which may help: https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning

PetricaM, 2018-09-04:
Setting a HA automation (that is MQTT, not UDP) which changes color (to a random color set) to run with 200 milliseconds delay (thus 5 Hz) works fine. Decreasing the delay to 100 ms generates erratic results (the gateway even reboots). 
Obviously, it depends on the gateway configuration (I've set mine in a three gateways network to ensure limited loss of packages). However, after 3 different UDP commands the gateway becomes unresponsive.  
Might need some different tuning to reach higher speed, but it is likely at the expense of reliability.

sidoh, 2018-09-10:
Going to close since it seems like the question was answered.  Please feel free to re-open if you want to continue the discussion.

closed 2018-09-10 by sidoh