sidoh/esp8266_milight_hub #306: B2 remote support

MiloshCZ, 2018-07-18:
Hi, there is no support for B2 remote.
[image]
ON button:
```
Packet received: 90F3381BD1CB4DBC17D1CEB2Packet transformed: 09CF1CD88BD3B23DE88B734DSuccessfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 53131
MiLightRemoteConfig::fromReceivedPacket: ERROR - tried to fetch remote config for unknown packet
WARNING: Couldn't find remote for received packet
Radio is available

Packet received: 90F3381BD1CB4DBC17Packet transformed: 09CF1CD88BD3B23DE8Failed CRC: expected 25491, got 15848
```
OFF button:
```
Packet received: 90AA9A609FB8005BF84CFF8CPacket transformed: 09555906F91D00DA1F32FF31Successfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 21810
MiLightRemoteConfig::fromReceivedPacket: ERROR - tried to fetch remote config for unknown packet

WARNING: Couldn't find remote for received packet
Radio is available

Packet received: 90AA9A609FB8005BF8Packet transformed: 09555906F91D00DA1FFailed CRC: expected 20352, got 55839
```
CH1 ON:
```
Packet received: 904998A7F442A85540469049Packet transformed: 099219E52F4215AA02620992Failed CRC: expected 3403, got 2450
Radio is available

Packet received: 904998A7F442A8554046D2B0Packet transformed: 099219E52F4215AA02624B0DSuccessfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 37474
MiLightRemoteConfig::fromReceivedPacket: ERROR - tried to fetch remote config for unknown packet
WARNING: Couldn't find remote for received packet
```
CH1 OFF
```
Packet received: 90FB3013DCC1160BE9EE85EEPacket transformed: 09DF0CC83B8368D09777A177Successfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 57207
MiLightRemoteConfig::fromReceivedPacket: ERROR - tried to fetch remote config for unknown packet
WARNING: Couldn't find remote for received packet
Radio is available

Packet received: 90FB3013DCC1160BE9EE85EEPacket transformed: 09DF0CC83B8368D09777A177Successfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 57207
```

MiloshCZ, 2018-07-18:
I think, that it uses new protocol: https://blog.christophermullins.com/2017/03/18/reverse-engineering-the-new-milightlimitlessled-2-4-ghz-protocol/

sidoh, 2018-07-22:
Support will be added in v1.8.  There's a feature branch with support for B2/FUT091:

https://github.com/sidoh/esp8266_milight_hub/tree/fut091_support

MisterC925, 2018-07-25:
Would the B2/FUT091, etc extend to the B4 and B8 remotes as well?

sidoh, 2018-07-25:
B4 should be RGB+CCT (FUT092).  B8 is the same as FUT089.  Both of these protocols are already supported.

MisterC925, 2018-07-25:
Sorry I assumed (poorly) that B2/4/8 would be related protocol wise (that'll teach me to think there's logic in any of this :) ) 

Thanks

sidoh, 2018-07-27:
haha.  not a ton of rhyme or reason in the model numbers.  :D

sidoh, 2018-07-29:
Support added in the 1.8 release.

closed 2018-07-29 by sidoh