sidoh/esp8266_milight_hub #307: Support of Touch Screen LED RGBW

sisamiwe, 2018-07-21:
Hi,
I ordered a MiLight RGBW controller with 2.4Ghz remote. I got this one, which seems not to be supported.
[image]
I cannot sniff pakets.
Can you help?
Thx

sidoh, 2018-07-22:
Think this is a duplicate of #266.

closed 2018-07-22 by sidoh

khmann, 2018-07-24:
Incorrect.  I own this module.  Notice how it has industry logos (CE something something) in the upper left where the word MiLight should be.  I have photos and SPI dumps.  It's a 6 byte protocol, it's actually pretty terrible, also the hardware is completely different inside.

coolduke666, 2018-11-20:
any solution for FUT027 support? milight oder clone? thx