sidoh/esp8266_milight_hub #310: Sniffing didnt find any thing

haymaan, 2018-08-03:
hello just installed the milight HUB and the webinter. looks very nice ,
i want to snif my MilightV6 bridge but no succes any help please ..?

P.S.   i havent change any think in settting !!
thanks

sidoh, 2018-08-03:
Hi @haymaan,

Please read through this troubleshooting guide: https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting

If you're still having issues, feel free to open another issue.

closed 2018-08-03 by sidoh

haymaan, 2018-08-03:
> esp8266_milight_hub/wiki/Troubleshooting

hello,
when i start the Snif and turn it on from the app(mobile) i see them in the
log but i can't add them?
when i poer up the light and click pair it's not pair them ?

sidoh, 2018-08-03:
ah, misunderstood.  yes, copy the ID field seen here:

[image]

into this field:

[image]

haymaan, 2018-08-03:
thanks for the reply,,,

my ID is

rgb_cct packet received (9 bytes):
Raw packet: B4 1F 20 3F AA 16 67 AA 6F

Decoded:
Key      : B4
b1       : 20
ID       : 0F28
Command  : 01
Argument : 06
Sequence : 99
Group    : 01
Checksum : F4

how can i convertit :) ?

sidoh, 2018-08-03:
you don't convert anything.  just type in "0x0F28".

haymaan, 2018-08-03:
okey ill try it thanks

haymaan, 2018-08-03:
sorry for my noob question,,,

but it dont save it :$ ?

how can i save it ..?

is there any way i can linky my milight Bridge V6 direct to it ... ?

sidoh, 2018-08-03:
You can save by opening the settings dialog and hitting "save."

They ought to auto-save, but this feature hasn't been added yet: https://github.com/sidoh/esp8266_milight_hub/issues/282

No, this is a replacement for a milight bridge.  It emulates the ID of the bridge.

haymaan, 2018-08-03:
thanks i'll play with it,,,
i want to stop useing controlicz and use philips emulator as its works
beter thats why i need it for alexa

thanks

haymaan, 2018-08-09:
please one more thing i have nodemcV3 but ive flash it with V2 software cuz
i can't find V3 software any where,is that will make problem ?iwant to give
it static IP where can i do that please ...?

thanks

>
>

sidoh, 2018-08-09:
Shouldn't matter -- the only real thing that matters for the versions is the size of the flash chip.  Quite a few of the firmwares are functionally equivalent, actually (ex, d1_mini and nodemcuv2).  Should be fine to use nodemcuv2 on a v3.

No way to set a static IP right now.  There's an open issue for it, though: #56

haymaan, 2018-08-09:
youy are a hero thanks,,, any knowloge how to get it work with Domoticz ive
the V6 ibox2 bridge working good with Domoticz but cant get this one work ?
thanks for your time ,sorry for my noob qeustions

sidoh, 2018-08-09:
I have no experience with Domoticz, so I probably won't be helpful.  But I know some folks have espMH working with it.  Maybe there are Domoticz forums or something where people have documented their setup?

http://www.domoticz.com/forum/viewtopic.php?t=18419 maybe?

looks like they're using the UDP gateways.

haymaan, 2018-08-09:
thank you already ask there waiting some answer still testing later this
month wil work on diyhue hope to link the milight hub and than to alexa
thanks friend