sidoh/esp8266_milight_hub #311: Same Sniff operation fails with 1.8 and not with 1.7.1

Brignoud, 2018-08-03:
Hi,
 I installed latest version, 1.8 for nodemecu and notice a strange behaviour
Sniffing operations seems not so responsive like 1.7
I’ve already remarked this with dev 1 and dev 2 but I was not so very sure but now a made a simple test with the official release with the order described below.
Tested with 1.7.1 and everything is right reported in the web interface
Installed the newest and operation doesn’t apply for so many time (nothing is sniffed)
Reverted again with previous 1.7.1 release and everything return to te correct efficient status

sidoh, 2018-08-05:
What remote are you sniffing?

I'm having a hard time reproducing this.  I'm getting 10s of packets per second on both 1.7.1 and 1.8.

Brignoud, 2018-08-06:
Basically FUT007 (I have a couple of this) both linked to the same bulbs (FUT017)
Not tested with B8, B4 and FUT092
I'll soon give it a try to confirm the same behavior

Brignoud, 2018-08-07:
Hi, 
I repeated the test including a B4 remote control also
I confirm the issue as:

with 1.81 (newly installed) sniff doesn't trigger anything
Reverted to 1.7.1 everything is correctly sniffed, the B4 also (with 3 FUT014 grouped all togheter)
Reinstalled 1.8.1 and encounterd the problem described (nothing is traced)
I have an openspace and both FUT014 and FUT017 are within the same space with the ESP Module  (30 Mt. environment) 
OpenHab2 on Raspberry system interact correctly and address any changes via the ESP Hub module
Can make a video if necessary with all the equipments and runtime test
Thank you

Brignoud, 2018-08-07:
Just to suggest any investigation, I've curiously found this on #266 here is an abstract:
----
sidoh commented on 31 May
Just an update here --

This is definitely an issue with the nRF radio code. I poked around a little bit, but haven't made much progress yet. (I did, however, manage to get the radio configs for all remotes supported by the ibox)

@khmann I recall that some of the radio optimizations you contributed a few months back may have broken sniffing from official devices (this was something we agreed was fine at the time). Does this sound familiar to you?

If so, any pointers on adding this support back without breaking the radio? Definitely happy to investigate myself, but thought I'd ask in case you have any ideas offhand.

 @sidoh sidoh added the bug  label on 31 May
 @sidoh sidoh changed the title from Newbie sniffing question to Packets from some remotes (e.g., RGB) are never sniffed on 31 May
@xbmcnut
xbmcnut commented on 5 Jun
Thanks for the update.

 @xbmcnut
xbmcnut commented on 14 Jun
Any NRF changes in the esp8266_milight_hub_nodemcuv2-1.8.0-dev1 build?

 @sidoh
Owner
sidoh commented 16 days ago
@xbmcnut hey, sorry I'm so slow getting back to you. Busy summer!

No, no nRF changes in v1.8. At the rate I'm going it probably won't make it in.

sidoh, 2018-08-08:
Yeah, some steps I could follow to reproduce this would be helpful.  I tried blasting my FUT092 (same thing as B4) with 1.8 and it was not noticeably slower than 1.7.1.

There aren't changes to radio code in 1.8.  That discussion is about certain types of remotes not being sniffable at all, and has been an issue for several versions.

sidoh, 2018-09-10:
Wondering if this is was the same problem reported in #327.  @Brignoud could you try 1.8.2?

sidoh, 2018-09-23:
Going to close this.  Please comment if you're still having trouble and we can re-open.

closed 2018-09-23 by sidoh