sidoh/esp8266_milight_hub #312: changing colors in HA not working in 1.8.0

EdgardosReis, 2018-08-04:
When I try to change the color in Home Assistant with the version 1.8.0 nothing happens. I have it configured as MQTT Lights and when I change the color a message like this is published:
{"state": "ON", "color": {"r": 0, "g": 0, "b": 255}}
However, if the bulb is already ON, the message is ignored. It seems that is ignoring the commands when the "state" is the same. If I publish a message like {"color": {"r": 0, "g": 0, "b": 255}} it works.
This doesn't happen with the version 1.7.1.

Best Regards.

sidoh, 2018-08-05:
Hi Edgar,

Can you tell me a little more about your setup?

* What remote type are you using? (RGB, RGB+CCT, etc)
* What group is your device bound to?
* What version of HASS are you on?

closed 2018-08-05 by sidoh

sidoh, 2018-08-05:
Fixed in 1.8.1, I hope (working for me, at least :) --

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.8.1

EdgardosReis, 2018-08-06:
Yes! Working now for me too. Thank you!

commit 7a15e234 (2018-08-05): Fix #312 -- multiple commands in same JSON would cause commands after 1st to get blank state
commit ea619b3c (2018-08-05): Merge pull request #313 from sidoh/312_fix_state-bug

Fix #312 -- multiple commands in same JSON would cause commands after 1st to get blank state