sidoh/esp8266_milight_hub #314: Home Assistant Control not fully working

nightcat91, 2018-08-06:
Hi,

today I was setting up the Milight Hub on my NodeMCU. Its working great but I found an issue together with HomeAssistant (HA).
I can control On and Off from HomeAssistant but changing brightness or color_temp does not work.

My setup:
Milight Hub on NodeMCU (software Version: 1.8.0 (nodemcuv2))
MQTT Server from HA (Hassio Addons, Mosquitto broker)
HA version: 0.75.1
MQTT topic pattern: espmilight/:device_id/:device_type/:group_id
MQTT state topic: espmilight/state/:hex_device_id/:device_type/:group_id

I am controlling CCT lights (led strips using the Milight controller), so I am using the Remote Type Fut091.
My device ID is 0x2810 ( I sniffed that from my existing hardware remote).

My HA config:
```
 - platform: mqtt_json
     name: "Küche Unterschrank"
     command_topic: espmilight/0x2810/fut091/1
     state_topic: espmilight/state/0x2810/fut091/1
     color_temp: true
     rgb: false
     brightness: true
```

Getting the state from the Milight Hub to HA works great, MQTT message looks like this:
```
Topic: espmilight/state/0x2810/fut091/1
Message: {"state":"ON","brightness":255,"color_temp":324}
```
HA updates the state corretcly.
Using HA to turn on and off the light works and send a mqtt message like this:
```
topic: espmilight/0x2810/fut091/1
message: {"state": "OFF"}
```
and this gets a state response message, all good.

but changing the color_temp inside HA triggers a mqtt message like this:
```
topic: espmilight/0x2810/fut091/1
message: {"state": "ON", "color_temp": 162}
```
But the light does not change, and I dont get a response.

Now with brightness its almost the same:
```
topic: espmilight/0x2810/fut091/1
message: {"state": "ON", "brightness": 61}
```
but now I get a state response (but its wrong):
Response:
topic: espmilight/state/0x2810/fut091/1
message: {"state":"ON","brightness":255,"color_temp":324}

One thing I found out:
letting the Milight Hub sniff while sending commands from HA resulted in these two packets beeing sniffed:
First Packet:
```
fut091 packet received (9 bytes):
Raw packet: 00 DC C9 13 66 CE D2 63 99 

Decoded:
Key      : 00
b1       : 21
ID       : 2810
Command  : 01
Argument : 02
Sequence : 78
Group    : 02
Checksum : 8E
```
Second packet: (they come right after each other)

```
fut091 packet received (9 bytes):
Raw packet: 00 DC E1 23 63 A3 D3 65 86 

Decoded:
Key      : 00
b1       : 21
ID       : 0000
Command  : 02
Argument : 3F
Sequence : 79
Group    : 00
Checksum : 93
```

Why does the second packet have ID: 0000???
Maybe thats why changing brightness and color_temp does not change anything on the actual lights??

If there is more info needed to track down the problem let me know!

EdgardosReis, 2018-08-06:
Try to update to version 1.8.1. I was getting the same problem.

sidoh, 2018-08-06:
Thanks a ton for the detailed debugging effort, @nightcat91.  Think this covers the same problem as #312.

Should be fixed in 1.8.1, as @EdgardosReis mentioned.

nightcat91, 2018-08-06:
Now I feel bad for not reading the patch notes.... sorry
but thanks for the quick help,, you are right 1.8.1 fixes the issue

sidoh, 2018-08-07:
Glad to hear that solves it.

No worries at all.  On the contrary, appreciate the level of depth in the bug report =)