sidoh/esp8266_milight_hub #315: Sending saturation to group 0 sets the bulb in white mode

PetricaM, 2018-08-08:
Hi Chris,

Sending a mqtt message (v. 1.8.1, but I think I've seen this also on previous versions) that includes also saturation to group 0 (either on rgb_cct or FUT089) gets the bulb first to red color then into white mode (the published state is the one intended however the bulb ends up in white mode).  There is no problem when sending the message to groups 1 to 4.

On a side note: remotes B0 and FUT088 are compatible and they both use FUT089 protocol (although with group 1; group 0 reacts as expected). Compared to four channels wall remotes (which are confusing for non-residents case a single bulb exists in the room :) ), B0 is also quite nice to be used as a dimmer beside a regular push switch or even as a replacement if push switch is only used seldom (case triggering lights by PIR sensors or by Google Assistant/Alexa). And, coupled with a home automation software (I use HA for some time), can even control other brands/technologies of light bulbs (even Philips Hue).

vansoest, 2018-08-19:
I can reproduce the problem, have exactly the same.
+1

sidoh, 2018-08-21:
I think this is a side-effect of the issue raised in #318. 

I would actually be a bit surprised if it affected both FUT089 and FUT092 (RGB+CCT).  FUT089 uses the same packet data for both kelvin and saturation commands, which are interpreted by the bulb depending on the state that they're in.  FUT092 uses different packet data for these two types of commands.

I think setting a color directly probably works for both (e.g., `{"color"{"r":255,"g":100,"b":100}}`) because the hue and saturation are computed and set at the same time.  But setting them in separate calls requires state to be kept.  And since state is not kept for group 0 it gets dropped.

Perhaps there's a more elegant solution to the group 0 issue that we haven't considered?  The facts I see are:

1. Group 0 isn't a group, it's a virtual wildcard group
2. Keeping state for Group 0 is misleading because state can also be set through commands sent to individual groups (ex. of confusing behavior: #284)

sidoh, 2019-05-19:
This should've been fixed with #426.

Pushed a test for it as well: https://github.com/sidoh/esp8266_milight_hub/commit/cb220e52fd3f5cddb1cb3709dc29284f397977e9

PetricaM, 2019-05-20:
Hi Chris,

It is fixed. 

Thanks

closed 2019-05-20 by PetricaM