sidoh/esp8266_milight_hub #317: Sniffer work paring the device ID not

poudenes, 2018-08-14:
Hi,

I can sniffer the bulbs ID. But when a add the ID and want pair them nothing happen.
All wire are correct connect. 

Firmware: 1.8.1
NodeMCU v3 with CH34* circled
NRF24L01+

sidoh, 2018-08-21:
Bulbs don't have IDs.  Do you mean that you can sniff a remote's ID?

Are the bulbs already paired with the remote?

poudenes, 2018-08-21:
Hmm I entered the ID that's shown in sniffer log but when I want control it nothing happens with the bulbs.

sidoh, 2018-08-21:
It sounds like a hardware defect.  Please read through this guide:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting

poudenes, 2018-08-30:
Checked it out. Got ID numbers. And was trying to test them. Suddenly find out that the given ID will switch on/off a total different bulb in the house. So send/receive work. But now have to find out what ID link to what bulb.

When i add a ID and refresh the page the added device id's are gone. IS that normal?

sidoh, 2018-08-30:
Depends on your definition of "normal," lol.

https://github.com/sidoh/esp8266_milight_hub/issues/282

shouldn't work that way, but currently does.

sidoh, 2018-09-10:
Going to close because it seems like this is resolved, but please feel free to reopen if that's not the case.

closed 2018-09-10 by sidoh