sidoh/esp8266_milight_hub #319: UDP v6 RGB+CCT color offset

amelchio, 2018-08-21:
Migrating from the Mi-Light iBox I notice that my RGB+CCT colors set with the UDP v6 interface have changed (this is with Home Assistant).

The hue range 0x00-0xff seems shifted by 8 so that e.g. 0xb2 is pure blue with the iBox while espMH needs 0xaa for the same color. I think this is a bug because the UDP interface is supposed to be compatible?

Furthermore, the hue 0x00 should be pure red with espMH but that message is actually ignored and the bulb does not change color at all.

I have dumped my test messages below. Getting these with Home Assistant was a bit tricky because it currently has a similar issues where the hue is shifted by 2. I have removed the top of a bulb to verify that these are indeed the values where only 1 LED is on.

**iBox**
```
RED
bytearray(b'\x80\x00\x00\x00\x11\x19\x02\x00\x07\x001\x00\x00\x08\x01\x08\x08\x08\x08\x04\x00^')

GREEN
bytearray(b'\x80\x00\x00\x00\x11\x19\x02\x00\n\x001\x00\x00\x08\x01]]]]\x04\x00\xb2')

BLUE
bytearray(b'\x80\x00\x00\x00\x11\x19\x02\x00\r\x001\x00\x00\x08\x01\xb2\xb2\xb2\xb2\x04\x00\x06')
```

**espMH**
```
IGNORED
bytearray(b'\x80\x00\x00\x00\x11\x00\x11\x00\x10\x001\x00\x00\x08\x01\x00\x00\x00\x00\x01\x00;')

RED
bytearray(b'\x80\x00\x00\x00\x11\x00\x11\x00\x0c\x001\x00\x00\x08\x01\xff\xff\xff\xff\x01\x007')

GREEN
bytearray(b'\x80\x00\x00\x00\x11\x00\x11\x00\x14\x001\x00\x00\x08\x01UUUU\x01\x00\x8f')

BLUE
bytearray(b'\x80\x00\x00\x00\x11\x00\x11\x00\x18\x001\x00\x00\x08\x01\xaa\xaa\xaa\xaa\x01\x00\xe3')
```

amelchio, 2018-08-24:
It seems that it was my test that was off by 2, not Home Assistant. I got confused because e.g. 0xb2 and 0xb4 appear indistinguishable.

sidoh, 2018-08-25:
Thanks for the detailed report!

Unrelated suggestion -- check out the [mqtt_json integration](https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant).  It works really nicely and has several advantages over UDP.

amelchio, 2018-08-25:
Right. I was missing the night mode when I tried out mqtt_json but I might move there eventually.

garmck, 2018-08-25:
@amelchio you can achieve night mode with mqtt as well using a mqtt_template light.

  _platform: mqtt_template
  name: Night Light
  state_topic: "milight/update/0x1/rgb_cct/1"
  command_topic: "milight/command/0x1/rgb_cct/1"
  command_off_template: '{"state":"off"}'
  command_on_template: '{"command":"night_mode"}'
  state_template: >
    {% if value_json.bulb_mode == 'night' %}
    on
    {% elif value_json.bulb_mode %}
    off
    {% endif %}_

The template can probably be optimised, but it works for me.

amelchio, 2018-09-05:
@garmck Thanks for the suggestion. However, to get a simpler setup, I wanted to make the night mode available as a Home Assistant light effect. That turned out to be rather easy because it is already implemented:

```yaml
light: 
  - platform: mqtt_json
    effect: true
    effect_list: ['night_mode']
    name: 'Office'
    command_topic: milight/0x1/rgb_cct/1
    state_topic: milight/states/0x1/rgb_cct/1
    color_temp: true
    rgb: true
    brightness: true
```

sidoh, 2018-09-10:
should be fixed in 1.8.2.

PR #320 (2018-09-09): Fix UDP v6 RgbCct color skew
This fixes #319 for me and UDP messages now set the same colors as my iBox.

I assume the call in the preset method would need to be similarly adjusted but I have no way of testing that at this time and I would like some feedback before spending more time on this.
closed 2018-09-10 by sidoh