sidoh/esp8266_milight_hub #323: Control bulbs using Domoticz

poudenes, 2018-08-30:
i have add all controllers (remote ID) into the WebUI as gateway. Add them into domoticz.

I see that the RGBW Bulbs working correct Color, sat, bright. But RGBWW change from color to Kelvin there is a little problem.

When change from Kelvin to a Color that is working. When you want select Kelvin again, it show the kelvin level with bulb but then it switch directly to the last selected color.

sidoh, 2018-08-31:
What remote type are you using?

Can you try disabling Automatic Mode Switching under Settings > Radio?

[image]

sidoh, 2018-09-10:
sweet 👍

closed 2018-09-10 by sidoh