sidoh/esp8266_milight_hub #324: Delete Device ID

poudenes, 2018-08-30:
Hi All,

Is there a way to delete Device ID's from the list? I was playing around and did some tests.
Now im know how to work with the web UI and want start with a clear module.

PetricaM, 2018-08-30:
I don't know if there's a better way, but you can save the settings on local drive (as a json file), edit (can add and remove from there any remote id), then load it back on the gateway.

poudenes, 2018-08-30:
Thanks that did the trick !!!

No have add 9 Device ID's and total 24 bulbs. Add them into Domoticz and now can control even bulb individual !!! GREAT WORK. Only see some problems with control bulbs color/Kelvin. Already open a issue.

sidoh, 2018-08-31:
Thanks @PetricaM!

There's not a way to do this through the UI.  You can either do as Petrica suggested, or use the `PUT /settings` endpoint, e.g. --

```
curl -X PUT -H 'Content-Type: application/json' -d '{"device_ids":[...]}'
```

note that you'll have to fully specify the list of device IDs you want to keep.

I believe there's another ticket out along these lines, so going to close.  But I'd agree with anyone suggesting this should be easier to do through the UI.

closed 2018-08-31 by sidoh