sidoh/esp8266_milight_hub #325: Empty /settings response

Esiravegna, 2018-08-30:
After updating mi nodemcuv2 from 1.6.5 to 1.8.1 I am effectively unable to save the mqtt settings, downloading the settings file, and if I set the mqtt settings, I'm no longer able to send responses to the bulbs.

Any ideas would be very much appreciated.

Esiravegna, 2018-08-31:
I got this in the log, as well:

```
Opening settings file failed
Restarting...
H⸮⸮⸮*⸮
J⸮Opening settings file failed
```

sidoh, 2018-08-31:
What's the output of `/about`?

Esiravegna, 2018-08-31:
```
{
version: "1.8.1",
variant: "nodemcuv2",
free_heap: 24088,
arduino_version: "2_4_1",
reset_reason: "External System"
}

Esiravegna, 2018-08-31:
I formatted the spiffs, and listed it and got an empty listing. I tought it may be lingering config files, but doesn't seems to be the case.

sidoh, 2018-08-31:
Good call.  I was going to suggest it sounds like SPIFFS got corrupted somehow, but you beat me to the punch.

What mechanism did you use to flash the new firmware?

Esiravegna, 2018-08-31:
I used a NodeMCU Flasher (windows 10, 64bits) and your bin images. It is possible that I haven't properly formatted the SPIFFS, I used SPIFFS.format(), do you know any other method?

sidoh, 2018-08-31:
Hm... I thought the pre-compiled binaries define the flash layout, but it's possible that the flashing tool does that as well.  Do you have the ability to select the size of your flash chip in the tool you used?

I think that works as well as anything.  There are some CLI tools as well (eg `esptool.py`).

It sounds like it worked, right?  I would expect an empty listing if you formatted it.  Does `/settings` return now?

Esiravegna, 2018-08-31:
No, it doesn't work :(, empty response :(. I'll try esptool

Esiravegna, 2018-08-31:
esptools worked!

Esiravegna, 2018-08-31:
THANKS A LOT.

closed 2018-08-31 by Esiravegna