sidoh/esp8266_milight_hub #326: Give Device ID a name

poudenes, 2018-09-01:
Not a issue, but a request:

I love the Milight Gateway Emulator for ESP 8266. Test it now with 32 bulbs and work great. Maybe a idea to make the WebUI better? Give Device ID a Name. So i add "0xAAAA" and want give this the name "Livingroom" When you press the pull down you get a nice list of all Device ID's with a nice name: Livingroom, Bedroom, Kitchen.. etc you name it.

Keep up the good work!!!

sidoh, 2018-09-01:
It's a good idea.

I think this is a duplicate of #210.  Does it look that way to you as well?

sidoh, 2018-09-10:
Thanks, gonna close as a duplicate.  Appreciate the suggestion.

closed 2018-09-10 by sidoh

poudenes, 2018-11-24:
Will this feature request be implemented in near future?

sidoh, 2018-11-24:
I have no active plans to do so.  As you can imagine, I work on this project in my spare time.  So new features get added when both time and interest permit. :)

More than happy to look at a PR!

poudenes, 2018-11-25:
We all more then happy you spend some of your time into this project!!
It was only a question :)

sidoh, 2018-11-25:
Totally fair question, I didn't mind :)