sidoh/esp8266_milight_hub #327: B04 group 0 is not 'sniffable' in 1.8.1

unbreakab1e, 2018-09-01:
Hi!

I've been running esp8266_milight_hub on nodemcuv2 since about may with homeassistant and today I decided to upgrade the firmware.
Suddenly, my esp8266_milight_hub <-> hass synchronization started to misbehave.
Upon investigation, i've determined that hass is no longer receiving any updates on ` milight/updates/0x364D/rgb_cct/0` topic, the "All' topic from my B4 remote.
Upon further investigation, I can tell that the hub is no longer able to sniff the press of that 'All' button.

I quickly assembled a second gateway, flashed it with 1.8.1, and tried sniffing with it, suspecting a faulty knockout nrf radio. Freshly built gateway sniffed group 0 fine.
Next, I downloaded the config backup from the second gateway, and compared it with the backup of the first one (taken it as a precaution before flashing the new version).

The diff was the following (passwords and ips redacted):
```
diff settings_sniff.json settings_nosniff.json
8c8
<         2
---
>         1
15d14
<         "mode",
27,29c26,28
<     "mqtt_password": "",
<     "mqtt_server": "",
---
>     "mqtt_password": "redacted",
>     "mqtt_server": "203.0.113.1",
31,35c30,34
<     "mqtt_state_topic_pattern": "",
<     "mqtt_topic_pattern": "",
<     "mqtt_update_topic_pattern": "",
<     "mqtt_username": "",
<     "packet_repeat_minimum": 4,
---
>     "mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
>     "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
>     "mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
>     "mqtt_username": "redacted",
>     "packet_repeat_minimum": 3,
38c37
<     "packet_repeats": 50,
---
>     "packet_repeats": 60,
41c40
<     "state_flush_interval": 10000
---
>     "state_flush_interval": 15000
```

I've uploaded the config from the first gateway into the second, and tried sniffing - no packets showed up.
So I started to adjust every option one by one until I was able to get it it sniff group 0 again, and it was MQTT stuff.
Namely, when i removed the mqtt server, group 0 packets started to show up again.

HenkDemper, 2018-09-03:
Hi Chris. After using the MiLight gateway 1.8.1 with MQTT for some time now I have 2 problems. The first one is as unbreakab1e: In the MiLight Hub Web GUI, I can also not see any packages for group 0 and (hence ?) also no MQTT update or state messages.

When I press Group 1-4 buttons on a remote, the Sniffing works quickly and I directly get the MQTT update and status commands. But when I press a group 0 On or Off button, then the sniffing stops (and doesn't recover in Web GUI, you need to refresh the browser) and no .../0 MQTT message is seen. In fact, the application seems to be frozen for about 12-15 seconds, after which it resumes to work (for group 1-4). This freezing also happens in the MQTT commands when you are not using the Sniffing mode in the Web GUI (or not having the Web GUI open at all). I tested many times and once I got a MQTT .../update/.../0 command.

When you goto the Settings menu in the Web GUI and make the MQTT server field empty, then do Start Sniffing group 0 commands are seen (but obviously nothing happens on MQTT any more), so same as unbreakab1e.

I was using group 0 update commands to change the state (keep in sync) of my items in openHAB automation, which now broke somehow after upgrading to 1.8.1 (from 1.8.0 Beta I believe)... Any fix would be welcomed. I have attached my (zipped) settings file would that help.

[settings.json.zip](https://github.com/sidoh/esp8266_milight_hub/files/2346448/settings.json.zip)

sidoh, 2018-09-03:
Is it possible this is a duplicate of #318 ?

HenkDemper, 2018-09-03:
Hi Chris. I'm not sure exact duplicate, but perhaps related. In my case, I'm never setting group 0 using MQTT ..../command/.../0, only (if it would work) trying to sniff when using external controllers like FUT006 and use MQTT .../update/.../0 to update my automation items.

Can you reproduce with my sample settings (or at least some MQTT server) and a remote ?

unbreakab1e, 2018-09-03:
Well, after reading through #318, and especially this comment - https://github.com/sidoh/esp8266_milight_hub/issues/318#issuecomment-417378901 , it looks like a duplicate.
It's just that my use-case is way more simple - I don't need any state sync between groups using "group zero", it's just a convenient option to tell homeassistant to kill the lights on the whole floor.

sidoh, 2018-09-03:
Curiously, I see this issue in the pre-compiled binary, but not on one I compile myself.  Probably some dumb uninitialized pointed or something.

unbreakab1e, 2018-09-04:
That’s strange because I compiled 1.8 binaries myself too. But the toolchain on my laptop is rather dated - hasn’t been updated since May or something.

sidoh, 2018-09-04:
Found the issue.  It was indeed a dumb thing.

Will need some time to refactor things.

Returning a reference to a local variable:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLightState/GroupStateStore.cpp#L28

HenkDemper, 2018-09-04:
> Returning a reference to a local variable:

Oops ;-) Glad you found it. Awaiting the fix and will confirm if it works then... Thanks !

sidoh, 2018-09-05:
potential fix in this PR: #330 

If this works I can release a patch version.  Would appreciate anyone who's able to test.  Here's a firmware image with the patch:

[firmware.bin.gz](https://github.com/sidoh/esp8266_milight_hub/files/2351921/firmware.bin.gz)

sidoh, 2018-09-10:
Should be fixed in v1.8.2.

PR #330 (2018-09-09): attempt fix for #327
This should hopefully fix #327 and maybe some related issues.
closed 2018-09-10 by sidoh